hardbyte/python-can #981: PCAN numerical vs string channel parameter

XelaRellum, 2021-02-28:
The implementation of the PCAN interface uses a string for the channel number.

This poses two problems:

- other interfaces use an int and end-user facing applications need to handle cases for the different interfaces (int for ixxat, vector, ..) and the symbolic name for _pcan_
- the actual value lookup is performed using _globals_ without checking the range of the name.

I propose 2 changes:

1. Allow an int to be used as channel parameter. the user of the function then needs to know the value, e.g. __0x51__ for _PCAN_USBBUS1_
2. Look up the channel value from a dictionary and this limiting this to valid channel names

felixdivo, 2021-04-14:
This seems to be a good addition. Would you mind creating a PR from [this](https://github.com/hardbyte/python-can/compare/develop...XelaRellum:pcan_numerical_channel)? We shouldn't need any deprecation notes for this.

felixdivo, 2021-04-14:
Already implemented in #982.

closed 2021-04-14 by felixdivo