hardbyte/python-can #986: Suppress Error or change encoding/language on PCAN interface

PierreLeger8, 2021-03-09:
I simulate some local nodes with this package, everything works fine but if my CAN-Bus goes down (e.g. Power down of NMT) i get following error messages in my terminal:
Bus-Fehler: Ein Fehlerz�hler hat die "heavy"/"warning"" Obergrenze erreicht bzw. �berschritten

In my pytest environment this messages breaks the html reporting (encoding error).
These three solutions would be acceptable for me:
1. Suppress this message
2. Get a correct encoded german message
2. Get the message in english

How can i reach one of this solutions?
Thanks for help

felixdivo, 2021-04-14:
Can you narrow down where the error is raised? Actually, this should work since pretty much everything in Python 3 is handled as Unicode. Could it be that pytest does not perform the right encoding / does it work from the console?

felixdivo, 2021-04-14:
Okay, actually a duplicate of #993.

closed 2021-04-14 by felixdivo