hardbyte/python-can #988: How to parse can frameID from the msg = bus.recv(1)

itsanjan, 2021-03-12:
I am using the following sample code to read CAN data from an ixxat device. 

`
def receive_all():

    bus = can.interface.Bus(bustype='ixxat', channel=0, bitrate=125000)

    # bus.state = BusState.ACTIVE  # or BusState.PASSIVE
    BusState.PASSIVE
    try:
        while True:
            msg = bus.recv(1)
            if msg is not None:
                print(msg)
                print(msg.data)
    except KeyboardInterrupt:
        pass
`

I am able to obtain the CAN data using msg.data however the documentation does not offer anything for the frameID's.
I am sure there is a native way, would be good if anyone can point me to the correct documentation. 👍 
Image for reference, i wish to extract the ID information.
[image]

KloolK, 2021-03-12:
That's the arbitration ID: https://python-can.readthedocs.io/en/master/message.html#can.Message.arbitration_id

itsanjan, 2021-03-12:
Thank you so much for the quick revert @KloolK . Cheers for the days !!

closed 2021-03-12 by itsanjan