hardbyte/python-can #992: send periodic a list of messages without wrapping

zorce, 2021-03-23:
Is it possible to send a list of messages periodically without wrapping when reaching the end of the list?
using the duration based on len(send_list) * cycle_time will overshoot and retransmit.

Creating a list of messages to send cyclic (including 20 or so muxed frames)

```
send_list = list()
for bmsg in bmsg_list:
    send_list.append(can.Message(arbitration_id=txid, is_extended_id=False, data=bmsg._data))

duration = len(send_list) * inhibit
task = c.bus.send_periodic(send_list, inhibit, duration=duration)
```

hartkopp, 2021-03-23:
So you would like so send this sequence as a one-shot event?
The **count** element in struct bcm_msg_head can limit the number of transmissions:
https://www.kernel.org/doc/html/latest/networking/can.html#broadcast-manager-protocol-sockets-sock-dgram
So if it is set to "20 or so" you should get a one-shot sequence transmission ...

zorce, 2021-03-24:
Yes that is correct, would like to send it as a one-shot event. Creating a list of muxed messages for "CAN-in-Automation" that needs to be sent on demand. 
If that count would be accessable as a parameter to send_periodic that would solve it for me just perfectly.

hartkopp, 2021-03-24:
https://python-can.readthedocs.io/en/master/bcm.html#can.broadcastmanager.MultiRateCyclicSendTaskABC is potentially what you need here. With subsequent_period being set to zero ...
But from the documentation I was not able to see whether you can provide more than one CAN frame via this API (for nframes being 20 or so)

zorce, 2021-03-24:
Have a fix for this, but have no idea how to push it :P
it is based on tag 4.0.0-dev.0 as I also need the send periodic with a list of multiple messages.