hardbyte/python-can #997: can.io.asc: not compatible with version 11.0.0

guillaume-pais, 2021-03-25:
The version 11.0 of CANalyzer uses an other format for the log files:

```
date Wed Mar 24 05:20:54.941 pm 2021
base hex  timestamps absolute
internal events logged
// version 11.0.0
Begin TriggerBlock Wed Mar 24 05:20:54.941 pm 2021
   0.000000 Start of measurement
   0.001317 CANFD   1 Rx        206  msg1                          1 1 f 64 16 6c c1 16 00 00 00 00 d8 82 2d 00 00 00 00 00 06 00 00 00 95 99 99 e9 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 40 00 00 00   152750  665   307000 a8098166 4b500150 4b100150 2000133a 2000030a
   0.001475 CANFD   1 Rx        205  msg2                 1 0 f 64 00 00 00 00 70 3d 0a 40 16 00 00 00 70 3d 0a c0 17 00 00 00 eb 7d 6d 3a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 40 00 00 00   152140  662   303000 d00d3475 4b500150 4b100150 2000133a 2000030a
   0.001574 CANFD   1 Rx        203  msg3                           1 0 d 32 01 00 00 00 00 00 ff ff 40 00 00 00 00 00 a5 1b 5b e4 7b 00 ef ff e9 ff 00 00 00 00 00 00 00 00    92468  363   303000 8003d8ca 4b500150 4b100150 2000133a 2000030a
   0.002013 CANFD   1 Tx        101  msg4                        0 0 b 20 21 00 05 00 0a 00 00 00 00 00 00 00 0a 00 ce ff 32 00 00 00   247203  250   301040 e00d5bb6 4b500150 4b100150 2000133a 2000030a
   0.002109 CANFD   1 Rx        203  msg5                           1 0 d 32 01 00 00 00 00 00 00 00 41 00 00 00 00 00 a5 1b 5b e4 7b 00 ef ff 00 00 00 00 00 00 00 00 00 00    92640  364   323000 c8011b0d 4b500150 4b100150 2000133a 2000030a
```

The old format is there: [CAN_Log_Trigger_3_2.asc](https://bitbucket.org/tobylorenz/vector_asc/src/47556e1a6d32c859224ca62d075e1efcc67fa690/src/Vector/ASC/tests/unittests/data/CAN_Log_Trigger_3_2.asc?at=master)

The solution could be to recognize the format in the header and then to parse differently depending on the version.

hartkopp, 2021-03-25:
Yes. Recent CANalyzer's support this new CANFD tagged format which can contain either CAN FD frames and Classical CAN frames.
You can find that information in the can_log_trigger_asc_format.pdf documantation from Vector.
The can-utils implementation (asc2log/log2asc) for CANFD is here:
https://github.com/linux-can/can-utils/blob/master/asc2log.c#L222

pierreluctg, 2021-03-30:
Is this not already in https://github.com/hardbyte/python-can/blob/develop/can/io/asc.py (develop branch)?

Anything missing?

felixdivo, 2021-04-23:
Maybe we could add a check [here](https://github.com/hardbyte/python-can/blob/0bcbf45d13375109ba654fbc4ab27041c1f435ab/can/io/asc.py#L84) which prints a warning if a version that is too new is detected.