hardbyte/python-can #998: Virtual Bus documentation example

polfeliu, 2021-03-25:
Hello, I'm using the virtual bus to test an application.

I learnt to use it with this example: https://github.com/hardbyte/python-can/blob/2.1.0/doc/interfaces/virtual.rst
I thought I had something wrong since the example raises an assertion error. Taking a closer look the messages "are not equal" because the timestamp and channels are different. The other data is identical and thus it works :). I would suggest to change the assertion of the example to comparison of data and dlc, or removing the assertion entirely as it may produce confusion.

polfeliu, 2021-04-01:
Oh! I didn't see that. Thank you! 
Readthedocs has the old example for some reason. https://python-can.readthedocs.io/en/master/interfaces/virtual.html
And the link to github points to the same example.

hardbyte, 2021-04-07:
Probably means we need a new release!

closed 2021-04-07 by hardbyte