hardbyte/python-can #1000: TypeError: cannot unpack non-iterable NoneType object running can_viewer.py against serial

truss-cpi, 2021-04-04:
I have an ADM3052 eval board outputting a raw bitstream via TTL/Serial. If I start can_viewer.py on Windows with no traffic present, it will start and show the blank table, but as soon as data starts coming through it crashes with the following details

`c:\Users\USER\AppData\Local\Programs\Python\Python39\Scripts>can_viewer.py -i serial -c COM5
Traceback (most recent call last):
  File "C:\Users\USER\AppData\Local\Programs\Python\Python39\Scripts\can_viewer.py", line 14, in <module>
    main()
  File "c:\users\USER\appdata\local\programs\python\python39\lib\site-packages\can\viewer.py", line 496, in main
    curses.wrapper(CanViewer, bus, data_structs)
  File "c:\users\USER\appdata\local\programs\python\python39\lib\curses\__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "c:\users\USER\appdata\local\programs\python\python39\lib\site-packages\can\viewer.py", line 74, in __init__
    self.run()
  File "c:\users\USER\appdata\local\programs\python\python39\lib\site-packages\can\viewer.py", line 84, in run
    msg = self.bus.recv(timeout=1. / 1000.)
  File "c:\users\USER\appdata\local\programs\python\python39\lib\site-packages\can\bus.py", line 83, in recv
    msg, already_filtered = self._recv_internal(timeout=time_left)
TypeError: cannot unpack non-iterable NoneType object`

I've observed the exact same error when running python-can under Linux in the same configuration. I don't really know python, but it seems like _recv_internal in bus.py doesn't actually return anything.

I was able to view some frames under SavvyCAN, so the data from the hardware would seem to be okay - though SavvyCAN is having it's own set of issues on both systems as well.

I'm it's user error, please let me know.

felixdivo, 2021-04-14:
Can you also reproduce this error on the develop branch? You should be able to install it with `pip install -e git+git://github.com/hardbyte/python-can.git@develop#egg=python-can[serial]`

truss-cpi, 2021-04-15:
I get the exact same behavior, yes

felixdivo, 2021-04-15:
Ahh. So I just added some typing annotations and found the problem. Indeed, if `rxd_byte != 0xBB`, the method will not return anything. The else-case is missing.

https://github.com/hardbyte/python-can/blob/cd87d2d86f44235a693338f22bf55198cea47879/can/interfaces/serial/serial_can.py#L151-L161

felixdivo, 2021-04-15:
Do you know what is sent on the serial bus when the error occurs? Maybe zero-length messages or text that is not a CAN frame?

felixdivo, 2021-04-15:
Is it intentional that at the very bottom of [this documentation](https://python-can.readthedocs.io/en/develop/interfaces/serial.html#internals), the "end of frame" is marked by `0xBBS` and not `0xBB`? That's not in line with either the code or the explanation above

felixdivo, 2021-04-15:
I improved the serial interface in #1010. It should print an exception with some debugging information. Could you install the version with `pip install -e git+git://github.com/hardbyte/python-can.git@small-simplifications-serial#egg=python-can[serial]` and paste the full exception message here? Please first check for sensible information, but there probably is non in it.

And thanks for debugging with me. :)

timothycpr, 2021-04-15:
I won't lie - it's entirely possible I'm not getting good data. But this is what I would be using to find that out, I don't have a scope or something similar to tell me if it's otherwise valid.

I'll re-run later and pass back what I get. If all we get out of this is an error telling us we don't have good data to work with, that's better than it throwing an exception and closing.

Thanks for taking a look.

felixdivo, 2021-04-15:
> Is it intentional that at the very bottom of [this documentation](https://python-can.readthedocs.io/en/develop/interfaces/serial.html#internals), the "end of frame" is marked by `0xBBS` and not `0xBB`? That's not in line with either the code or the explanation above

@hardbyte @boris-wenzlaff Do you know what's up with the docs here? You two seem to have touched this the last time.

truss-cpi, 2021-04-15:
Here's what I got
`Traceback (most recent call last):`
`  File "/usr/local/bin/can_viewer.py", line 7, in <module>`
`    exec(compile(f.read(), __file__, 'exec'))`
`  File "/usr/local/lib/python3.8/dist-packages/src/python-can/scripts/src/python-can/scripts/can_viewer.py", line 11, in <module>`
`    main()`
`  File "/usr/local/lib/python3.8/dist-packages/src/python-can/scripts/src/python-can/can/viewer.py", line 556, in main`
`    curses.wrapper(CanViewer, bus, data_structs)`
`  File "/usr/lib/python3.8/curses/__init__.py", line 105, in wrapper`
`    return func(stdscr, *args, **kwds)`
`  File "/usr/local/lib/python3.8/dist-packages/src/python-can/scripts/src/python-can/can/viewer.py", line 72, in __init__`
`    self.run()`
`  File "/usr/local/lib/python3.8/dist-packages/src/python-can/scripts/src/python-can/can/viewer.py", line 82, in run`
`    msg = self.bus.recv(timeout=1.0 / 1000.0)`
`  File "/usr/local/lib/python3.8/dist-packages/src/python-can/scripts/src/python-can/can/bus.py", line 87, in recv`
`    msg, already_filtered = self._recv_internal(timeout=time_left)`
`  File "/usr/local/lib/python3.8/dist-packages/src/python-can/scripts/src/python-can/can/interfaces/serial/serial_can.py", line 173, in _recv_internal`
`    raise RuntimeError(`
`RuntimeError: TODO, see issue #1000; locals = {'self': <can.interfaces.serial.serial_can.SerialBus object at 0x7f9e5b65c070>, 'timeout': 0.0007379779815673828, 'rx_byte': b'\xaa', 's': b'\xb7\xd8\xd4\x11', 'timestamp': 271510418, 'dlc': 116, 'arb_id': 299161783, 'data': b'\x908H\x12\x0f\x92\xf8\xb9\x82\xfa\x11\x10\xc0\x13\x10\xc8\x9e\x10\xfc\xa1\x15\xe9!\x00\xf7\x040\xf4Z\x80\xfb\x01"\xd6#\x00\xf4#\x10\xe4\x11\x05\xe6\x0c\xb4=\xa0W.\x00t\xb7\xd8\xd0\x11\x90?H\x12\x0f\x92\xdc\x11\x10\xe4\x13\x10\xecP\n\xfaD\x89\xfe\n$\xc0 0\xe6\xbc@\xbf\x01"\xfe\x01"\xd6#\x10\xf4\x03\x10\xc4\x11\x05\xee\x08\xdc?\x89\xef.\x10\xac\xa1\x15\xe9x\xe2}#\xb0O\x80', 'rxd_byte': 128}`

felixdivo, 2021-04-16:
The `locals()` were:

```
{
   "self": <can.interfaces.serial.serial_can.SerialBus object at 0x7f9e5b65c070>,
   "timeout": 0.0007379779815673828,
   "rx_byte": b"\xaa",
   "s": b"\xb7\xd8\xd4\x11",
   "timestamp": 271510418,
   "dlc": 116,
   "arb_id": 299161783,
   "data": b"\x908H\x12\x0f\x92\xf8\xb9\x82\xfa\x11\x10\xc0\x13\x10\xc8\x9e\x10\xfc\xa1\x15\xe9!\x00\xf7\x040\xf4Z\x80\xfb\x01"\xd6#\x00\xf4#\x10\xe4\x11\x05\xe6\x0c\xb4=\xa0W.\x00t\xb7\xd8\xd0\x11\x90?H\x12\x0f\x92\xdc\x11\x10\xe4\x13\x10\xecP\n\xfaD\x89\xfe\n$\xc0 0\xe6\xbc@\xbf\x01"\xfe\x01"\xd6#\x10\xf4\x03\x10\xc4\x11\x05\xee\x08\xdc?\x89\xef.\x10\xac\xa1\x15\xe9x\xe2}#\xb0O\x80"
   "rxd_byte": 128
}
```

felixdivo, 2021-04-16:
I suspect that the "messages" you receive are broken. The DLC is `116`, which is way beyond the maximum of `8` that CAN frames can hold. I'll implement a check for the validity of the DLC and raise an error about invalid received data in a case where the end delimiter is missing.

Sorry, but I probably cannot help you much further since only you know the specifics of your setup. The `rx_byte` and `timestemp` look reasonable, btw. However, the arbitration ID is also out of range.

timothycpr, 2021-04-16:
More help isn't really necessary - I was using this setup to try and capture because it required me to steal the least amount of hardware from other projects.

I moved on to flashing my CANDue and have been successful capturing from that into SavvyCAN, but figured I'd do what little I could with passing this error info back.

felixdivo, 2021-04-16:
Got it. And thanks for reaching out! This PR will be closed once the improved error messages in #1010 get merged.

PR #1010 (2021-04-18): Improve code of serial interface
This PR improves the serial interface:

- add typing annotations
- simplify code
- some more inline docs
- better validation of received messages / more robust (now throws exceptions on faulty data)
- fixes #1000
- fixes [this](https://github.com/hardbyte/python-can/runs/2351273197?check_suite_focus=true#step:5:393) mildly related error, closes #1014
closed 2021-04-18 by felixdivo