hardbyte/python-can #1001: Message.data returning byte array not decoded data

humishum, 2021-04-06:
Python Version: 3.6

When calling for the .data attribute on a message, I'm getting returned a byte array. When decoding this byte array(with UTF-8 as specified) I get a "codec can't decode byte 0x__ in position __: invalid start byte" UnicodeDecodeError. I've tried digging in and unpacking the struct itself, getting an error each time with a buffer size mismatch. Is there a way to decode the byte array from message.data ? 

Using blfReader() to load a blf file, and extracting data from messages with an iterator through log 

```
log = can.BLFReader(blfFile)
 
log_output = []

for msg in log:
    log_output.append([msg.timestamp, msg.arbitration_id, msg.id_type, msg.channel, msg.dlc, msg.data] )
```

msg.data returns a byte array
msg.data.decode() returns a UnicodeDecodeError as stated above. 

Should also be noted when I call msg.equals the data is stored in an array of hex numbers, not a byte array, so how can I access this?
msg.equals returns <bound method Message.equals of can.Message(timestamp=1613835116.766218, arbitration_id=0x2a5, extended_id=False, channel=5, dlc=2, data=[0x6c, 0x25])>

hardbyte, 2021-04-06:
Are you expecting the binary data to be Unicode? That would be unusual so I'd first suggest checking that assumption.

The Message string representation shows the `data` as hex for convenience - see the implementation [here ](https://github.com/hardbyte/python-can/blob/develop/can/message.py#L178).

humishum, 2021-04-07:
Yeah I read through the implementation and somehow concluded that `.data` would give me back the values in hex rather than a byte array, which I now see is wrong. My apologies for that! 

I needed the values in hex to output to csv for readability, so found a workaround for that by parsing the string, but the issue with decoding the byte array from the Message class still stands. From my understanding, the data is encoded/decoded in standard UTF-8, but when running 'data.decode('utf-8') I encounter that UniCodeDecodeError as mentioned earlier.

hardbyte, 2021-04-07:
You shouldn't need to parse strings, the `data` is easily converted to a list of hex strings with `["{:#02x}".format(byte) for byte in msg.data]`.

> the issue with decoding the byte array from the Message class still stands. From my understanding, the data is encoded/decoded in standard UTF-8, but when running 'data.decode('utf-8') I encounter that UniCodeDecodeError as mentioned earlier.

Could you expand on where the data comes from and why you're expecting it to be utf-8? What are you trying to achieve? 
Again I'll point out it would be unusual for the data transmitted in a can frame to be utf-8.

felixdivo, 2021-04-23:
It is not clear how this library should be changed due to this issue. If there are further problems feel free to open a new issue.

closed 2021-04-23 by felixdivo