hardbyte/python-can #1002: Hardware timestamps for can.Bus()

stefanbie, 2021-04-08:
Hi!

I have a Kvaser dongle connected to my machine where I listen to all messages och a channel and log the timestamps. So far I have used  "candump -ta -H can0" for this and it works fine. I want to log hardware timestamps as time is crucial (why I use -H for candump) The output looks like this:

(0000775431.709973)  can0  702   [8]  10 03 02 02 00 16 27 DC
(0000775431.759950)  can0  702   [8]  18 04 02 00 31 11 5E AA
(0000775431.835127)  can0  702   [8]  10 03 03 02 00 16 27 DC

Now, I want to use python-CAN for this instead. I have used following code to listen to a CAN channel. 

```
import can

bus = can.Bus(channel='can0', bustype='socketcan',bitrate=500000)
reader = can.BufferedReader()
notifier = can.Notifier(bus, [reader])

while True:
    msg = reader.get_message()
    print(msg)
```

Now, I get the following output:

Timestamp: 1617866119.201752        ID: 0702    S                DLC:  8    18 04 0b 00 38 85 92 04     Channel: can0
Timestamp: 1617866119.276675        ID: 0702    S                DLC:  8    10 03 0c 02 00 16 24 70     Channel: can0
Timestamp: 1617866119.326673        ID: 0702    S                DLC:  8    18 04 0c 00 04 5d 3a 18     Channel: can0

Obviously the timestamp is not the same as in candump so I suspect that this is sw timestamp, is that coorect?. In that case, is there some way to specify that I would like to use hardware timestamp instead?

BR Stefan

christiansandberg, 2021-04-08:
It is a hardware based timestamp but we convert it to Unix time so it is the same as for other interface implementations.

stefanbie, 2021-04-08:
Hi Christian! Thanks for the reply and great that it is hw timestamps by default. May I ask how it is converted?

christiansandberg, 2021-04-08:
When the interface is initialized both the hardware timer and the host's absolute time is read. These are then used to calculate the offset which is added to the timestamp of the messages.

You can look at the implementation for more details:

https://github.com/hardbyte/python-can/blob/develop/can/interfaces/kvaser/canlib.py

stefanbie, 2021-04-09:
Thanks a lot for the clarification! :)

felixdivo, 2021-04-14:
These does not seem to be a clear action to be done here, so I'm closing this for now. If there is need for updating docs or code, please comment here or open a new issue.

closed 2021-04-14 by felixdivo

stefanbie, 2021-05-17:
Hi, @christiansandberg, I had another comment on this :) According to to the hardware manufactory TX hardware timestamps is not supported by socketcan, is that the fact?

christiansandberg, 2021-05-17:
That I don't know, but if you set `receive_own_messages=True` in the bus constructor you should get your sent messages back with some kind of timestamp at least.

stefanbie, 2021-05-17:
Ok, thanks for the feedback!