hardbyte/python-can #1008: Create Issue Templates

felixdivo, 2021-04-14:
I just closed more than a dozen stale issues. Quite a few were not reproducible, and having known the users OS, Python and library version involved would've helped a lot. I think that adding templates for issues would've caused users a tiny bit more effort but would help the maintainers a lot.

It would look like this when creating new issues (see the [GitHub docs on these templates](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository)):

[image]

I'd propose two issue templates: "Bug Report" and "New Feature". We can still leave the option to start with a blank issue. We might consider a simple PR template too, such that no one forgets to close and issue with `Close #007` or `Fixes #007`. Something simple like [this](https://github.com/stevemao/github-issue-templates/tree/master/simple) would probably already suffice.

hardbyte, 2021-04-15:
Sounds good to me! 

Suggest something like this on the end of the issue template:
  

  **Additional context**
  
  
  
  <details><summary>Traceback and logs</summary>
  
  
  
  
  
  ```python
  def func():
      return 'hello, world!'
  ```
  </details>

felixdivo, 2021-04-15:
Sounds good.

According to the link to the docs above, only someone with admin rights to the projects settings may change the templates. So that's maybe only you, @hardbyte.

hardbyte, 2021-04-15:
That's not my understanding, they are just files in the repo?

felixdivo, 2021-04-18:
It's easy to add them here: https://github.com/hardbyte/python-can/issues/templates/edit

PR #1017 (2021-04-21): Add Bug report issue template
Closes #1008
closed 2021-04-21 by hardbyte