hardbyte/python-can #1021: No "is_rx"  property in message

stefanbie, 2021-04-21:
Hi 

Im running python-can 3.3.4 with python 3.8.5. When I run the code below to listen to messages on can0...

`
bus = can.Bus(channel='can0', bustype='socketcan', bitrate='500000')
reader = can.BufferedReader()
notifier = can.Notifier(bus, [reader])
msg = reader.get_message(timeout=100)
`
... the message revived looks like this:

[image]

There is no "is_rx" property as stated in the documentation, how can this be achieved? 

BR Stefan

zariiii9003, 2021-04-21:
It is available on the develop branch. You could try the pre release version [here](https://github.com/hardbyte/python-can/releases/tag/4.0.0-dev.1)

stefanbie, 2021-04-22:
Thanks for your answer, very appreciated! Now I can see the Rx/Tx *property, however, timestamps seems to be in seconds by default?

felixdivo, 2021-04-22:
Timestamps are always in seconds, yes. This is normal for Python, e.g. see [`time.time`](https://docs.python.org/3/library/time.html#time.time).

stefanbie, 2021-04-22:
I believe, In 3.3.4 default timestamp precision is microseconds, at least for hw timestamp that I use. Now with dev branch I get second precision.

felixdivo, 2021-04-22:
Even in your version, the timestamp should've been in seconds. See [the docs](https://python-can.readthedocs.io/en/3.3.4/message.html#can.Message.timestamp).

stefanbie, 2021-04-22:
With the same code as above I get following message with the dev (4.0.0.dev0) branch:

[image]

See the difference in timestamp?

felixdivo, 2021-04-22:
Oh sorry! You meant the *precision*, not the format.

I can also reproduce it with this on commit 404876533b2ec886732869b8a8756018907acae9 (currently the most recent one on `develop`)

```python
import can

with can.Bus(channel='vcan0', bustype='socketcan') as bus:
    with can.Bus(channel='vcan0', bustype='socketcan') as bus_sender:
        bus_sender.send(can.Message(arbitration_id=3))

    print(bus.recv(timeout=100))
```

felixdivo, 2021-04-23:
You can now install version `4.0.0.dev2`.

stefanbie, 2021-04-26:
Hi

I tried to use 4.0.0.dev2 (wget
https://github.com/hardbyte/python-can/archive/refs/tags/4.0.0-dev.2.tar.gz)
but it was still seconds precision.

BR Stefan

felixdivo, 2021-04-26:
Yes, that's because #1029 is not merged yet and also not release. You can install from that branch to get the fix immediately. Apparently, something like this should work: `pip install https://github.com/user/repository/archive/branch.zip`.

PR #1029 (2021-05-10): Fix size/layout of struct needed for time stamping
This caused the sub-second resolution to be cut off.

Fixes #1021. Also adds a regression test to catch this in the future and with other interfaces too.
closed 2021-05-10 by mergify[bot]