hardbyte/python-can #1039: Notifier fails silently when using Listeners

sdorre, 2021-04-30:
It may be a mistake on my side. But I wanted to double check.

**Describe the bug**
When using the Notifier with the default `_rx_thread` processing thread, if an exception occurs, it tries to call `on_error()` from all the Listeners. 
As the classes all inherits from `class Listener(metaclass=ABCMeta):`
This class has already defined `on_error` But does not do anything.
```
    def on_error(self, exc: Exception) -> None:
        """This method is called to handle any exception in the receive thread.

        :param exc: The exception causing the thread to stop
        """
```

So in the Thread the error handling never says anything. And Error is not displayed anywhere. 
```
        except Exception as exc:  # pylint: disable=broad-except
            self.exception = exc
            if self._loop is not None:
                self._loop.call_soon_threadsafe(self._on_error, exc)
                raise
            elif not self._on_error(exc):
                raise
```

This is an example class:
```
class CustomListener(Listener):

    def on_message_received(self, msg: Message):
        raise Exception("What should we do now?")

    def stop(self):
        self._log_file.close()
```

So if that is not a bug, what should the `def on_error(self, exc: Exception):` look like?

felixdivo, 2021-05-02:
I guess that such a Listener would do the job of crashing the thread:

```python
class ErrorCrashListener(Listener):

    def on_error(self, exception: Exception):
        raise exception
```

The API is indeed a bit strange here. If you do not override it, error handling defaults to doing nothing (the method as implemented in the base `Listener`). Not even the exception is logged. But also, the tread continues to listen for messages. I changed that in #1040.

felixdivo, 2021-05-04:
@sdorre Does that answer your question or did it create more? ;)

sdorre, 2021-05-06:
No that is right. I should simply raise it by myself. Thanks for the hint!