hardbyte/python-can #1041: Add documentation page on related tools

felixdivo, 2021-05-05:
It is crucial to limit the scope of the `python-can` library to keep the maintenance effort manageable. However, many users will probably look in the docs of this library seeking some related functionality. It could be useful to add a page on such related tools.

Examples of ideas that came up
- *Add support for Lin to Blf Reader* #1037
- *viewer.py with options to publish scaled & labeled CAN signals to MQTT* #967

zariiii9003, 2022-11-17:
I like this, this could help reduce new issues in this repo. I would suggest mentioning popular libraries for common problem domains like

- J1939
- ISO-TP
- DBC/ARXML
- UDS
- CANopen
- XCP

Of course the mentioned libraries should be actively maintained and have python-can as a dependency.

hardbyte, 2022-11-17:
I'm a fan of this idea too, I wouldn't restrict it to things using python-can e.g. could include:

- directly using python `sockets`
- Micropython's CAN api

felixdivo, 2022-11-20:
I also think that mentioning some external tools and alternatives would be useful. Someone has to write it though ;)

closed 2022-12-19 by zariiii9003