hardbyte/python-can #1053: pcan timestamps are silently incorrect on windows, (without uptime installed)

mjfwest, 2021-05-19:
**Describe the bug**
A fresh install of python-can, using a pcan device results in incorrect timestamps as `uptime` is not a requirement and the code below is designed to fail silently.
```python
try:    # use the "uptime" library if available
    import uptime
    import datetime
    # boottime() and fromtimestamp() are timezone offset, so the difference is not. 
    boottimeEpoch = (uptime.boottime() - datetime.datetime.fromtimestamp(0)).total_seconds()
except ImportError:
    boottimeEpoch = 0
```
**To Reproduce**
any code as described above

**Expected behaviour**
Raise the exception, or give a warning that the timestamps will be incorrect.

**Additional context**

OS-version: Windows
Python version: 3.9
python-can version: '3.1.0'
python-can interface/s: PCAN

  
<details><summary>Traceback and logs</summary>

  

  

```python
def func():
    return 'hello, world!'
```
</details>

mjfwest, 2021-05-19:
of course manually installing uptime solves this.

felixdivo, 2021-06-02:
As it's [just 6.6 kB](https://pypi.org/project/uptime/3.0.1/#files), we might add it as a `python-can[pcan]` addition.

PR #1093 (2021-06-15): Improve PCAN installation procedure
Fixes #1053.

This PR:
- prints a warning is `uptime` is missing in PCAN
- Adds a `setup.py`-extra to be able to easily install it
- Changes the documentation accordingly
- Fixes some small documentation issues like old links, non-HTTPS links and formatting
closed 2021-06-15 by felixdivo