hardbyte/python-can #1057: sysWORXX, Systec CANmodul8, open both channels on one physical device

KB20236, 2021-05-28:
On Windows it is not possible to create 2 bus objects and open the interface for both channels on one physical device, like it is in the code example below.
 
```
bus1 = can.interface.Bus(bustype="systec", device_number="0", channel="0", bitrate=500000)
bus2 = can.interface.Bus(bustype="systec", device_number="0", channel="1", bitrate=500000)
msg1 = bus1.recv(timeout=2)
msg2 = bus2.recv(timeout=2)
print(msg1, "\n", msg2)
bus1.shutdown()
bus2.shutdown()
```

I think the problem is, that whenever a new bus is initialized, the UcanInitCanEx() function from the USBCAN32.dll is called. This function does not have a channel parameter. To open a second channel on the same device the UcanInitCanEx2() has to be called.

**CANRET PUBLIC UcanInitCanEx (tUcanHandle UcanHandle_p, tUcanInitCanParam* pInitCanParam_p);

UCANRET PUBLIC UcanInitCanEx2 (tUcanHandle UcanHandle_p, BYTE bChannel_p, tUcanInitCanParam* pInitCanParam_p);**

felixdivo, 2021-06-29:
@KB20236 We always welcome contributions! This looks useful.

zhoulian1992, 2022-05-16:
> @KB20236 We always welcome contributions! This looks useful.

Has this problem been solved?

zhoulian1992, 2022-05-16:
Every time I call the can.interface.Bus() fuction, the init function will be used. The value hw_is_initialized and ch_is_initialized are been set to False. In the second time I call the can.interface.Bus() ,the UcanInitHardwareEx() will be also called. Then the error is occurred.

felixdivo, 2022-05-16:
(I won't work on this issue, btw. I just commented above.)

peterk62, 2023-10-10:
I've come accross the same problem today. I've got a very recent install, so it looks like it hasn't been fixed yet.

The problem (as was noted above) is that the initialise method does not have persistance between calls that the hardware has already been initialised by another call, nor does it keep track of the needed device handle from the first call for subsequent operations. I've made some crude changes to the ucanbus.py file to make the hw_init flag and the handle global and it works a treat. I can not use both channels. Not tested for multiple devices - most likely will not work as I have made no provision for it.

I've attached the file here in case it helps anyone else out. Not sure how to propose the change for incorporation. It's pretty crude as my Python skills are rusty. Any assistance/comments appreciated.

[ucanbus.zip](https://github.com/hardbyte/python-can/files/12862806/ucanbus.zip)