hardbyte/python-can #1058: usb2can interface error

ivanrulik, 2021-05-28:
**What is the problem**
I'm trying to connect with a CAN debugger (zeroerr company) who uses USB2CAN Interface but I only the error [see section below] 

**Expectations**
I was expecting that the code will be able to fin the CAN debugger connected

**Additional context**

OS-version: Windows 10
Python version: Python 3.7.9 32 bit
python-can version: 1.2.1
python-can interface/s: usb2can

  
<details><summary>Code</summary>

 

```python
import canopen
import os
import time
node = canopen.RemoteNode(1, "Driver_V1.1.eds")
network.add_node(node)
network.connect(bustype='usb2can')
...
network.sync.stop()
network.disconnect()
```
</details>

<details><summary>Error</summary>

 

```python
Traceback (most recent call last):
  File "c:/Users/user/Desktop/Python/canopenTest.py", line 17, in <module>
    network.connect(bustype='usb2can')
  File "C:\Users\user\AppData\Local\Programs\Python\Python37-32\lib\site-packages\canopen\network.py", line 109, in connect
    self.bus = can.interface.Bus(*args, **kwargs)
  File "C:\Users\user\AppData\Local\Programs\Python\Python37-32\lib\site-packages\can\interface.py", line 125, in __new__
    return cls(*args, **kwargs)
  File "C:\Users\user\AppData\Local\Programs\Python\Python37-32\lib\site-packages\can\interfaces\usb2can\usb2canInterface.py", line 103, in __init__
    raise CanError("could not automatically find any device")
can.CanError: could not automatically find any device
```
</details>

felixdivo, 2021-06-05:
cc @christiansandberg

holmel, 2021-06-11:
Changing this line 

https://github.com/hardbyte/python-can/blob/8e6f05a1e9075da13fa52c900cc0be3e3f7d7833/can/interfaces/usb2can/serial_selector.py#L42

to
```
def find_serial_devices(serial_matcher=""):
```
solves this same issue  for me

felixdivo, 2021-06-15:
> I think the proper way is to provide the serial number when connecting:

Thanks @holmel. Yes, that's correct. They way, it should always work. @ivanrulik Does that solve you issue?

ivanrulik, 2021-06-15:
Yes, it did solve the issue. Thanks!

closed 2021-06-15 by ivanrulik