hardbyte/python-can #1059: Upload Python Wheels to PyPI

seifertm, 2021-05-31:
**Is your feature request related to a problem? Please describe.**
It is currently very difficult to install or package `python-can` for platforms different than the platform of the current build host. For example, on a Linux x86_64 machine, there is no easy way to install `python-can` for Linux armv7 or any other platform.

The main reason for this problem is that `pip install --platform=…` [requires all dependencies to be available as binary Wheels](https://pip.pypa.io/en/stable/cli/pip_install/#cmdoption-platform) for the target platform. However, the most recent stable version of `python-can` (v3.3.4) [only provides a source tarball](https://pypi.org/project/python-can/3.3.4/#files) on PyPI.

The problem stretches a bit further than just `pip install`, though. Users cannot easily create a [zipapp](https://docs.python.org/3/library/zipapp.html) archive from applications using `python-can` that target a different platform. Nor can they use tools that rely on Pip to create cross-platform compatible executables, such as [PEX](https://pex.readthedocs.io/en/v2.1.42/).

**Describe the solution you'd like**
A Python Wheel is provided for each platform supported by `python-can` for every future release.

**Describe alternatives you've considered**
Manually build a Python Wheel for the target platform and host it on a private package index. Drawbacks:
* Requires a build setup for the target platform
* Requires private package index to be maintained
* Requires repeated work when updating python-can to a new version

**Additional context**
A universal Wheel was provided until [python-can v3.1,](https://pypi.org/project/python-can/3.1.1/#files) but Wheel support seemingly has been dropped starting from python-can v3.2. Development builds of python-4.0.0 [seem to provide](https://pypi.org/project/python-can/4.0.0.dev2/) an .egg file in PyPI, but no Wheel.

Based on my current understanding, python-can is platform independent, so a single universal Wheel would cover all supported platforms and the corresponding interpreters. Is this correct?

hardbyte, 2021-06-01:
We aim to build and publish a wheel when releasing - see the release guidelines in https://python-can.readthedocs.io/en/develop/development.html#creating-a-new-release

felixdivo, 2021-06-02:
I will have to delay the release of 4.0.0 by a week due to the combination of good weather and public holiday. ;) But I'll read up on the issue next week, thanks for bringing it up @seifertm.

BTW: At least the most recent 4.0.0 development release was generated automatically by [`.travis.yml`](https://github.com/hardbyte/python-can/blob/develop/.travis.yml#L84-L95). According to [the docs of Travis](https://docs.travis-ci.com/user/deployment/pypi/#uploading-different-distributions), one could specify `distributions: "sdist bdist_wheel"`. Would that already solve the issue?

closed 2021-06-08 by mergify[bot]

seifertm, 2021-06-16:
Thanks @felixdivo and @hardbyte for taking care of this so quickly!

PR #1071 (2021-06-08): Add bdist_wheel distribution to automated Travis CI deployment
Fix #1059 for the next release. This will be tested on the next release candidate.