hardbyte/python-can #1060: long time logging in stepped blf files!

GitYom, 2021-05-31:
Hello,

I was interesting in logging messages from two can buses into blf files for a long time period (6 to 20hours a day).
the blf files should be sent by 4G to a server.
I was thinking of logging messages in small files (less than 1Mbytes).
My first idea to do this is to use the BufferedReader  as in the following code.
What do you think about my idea? Do you think it could work in an efficient way? or Is there a simple and efficient way to do it?

Best Regards,

Guillaume.

--------------------------------------------------------------------------------
'import can
! continuous logging messages from two can buses in blf file for long period
! to avoid heavy blf file, a new indexed blf file is created when the max number of messages is reached
! use of BufferedReader to this purpose
! two can buses on my interface
bus0 = can.interface.Bus(channel='can0', bustype='socketcan')
bus1 = can.interface.Bus(channel='can1', bustype='socketcan')
! bufferedReader to queue all incoming messages
buffer = can.BufferedReader()
notifier_1 = can.Notifier([bus0, bus1], [buffer])
! file index to record
index = 0
! number of messages to record in blf file
max_messages = 1000
! time to wait a new message
time_out = 2
!external signal to stop the recording process
stop_demand = False
while not stop_demand:
	index +=1
	messages = 1
	! new file is created according to index and blf reader is attached to it
	file_out = "fichier_{:0>6d}.blf".format(index)
	blf_writer =  can.BLFWriter(file_out)
	! while the max number of messages is not reached, we write in the blf file
	while messages <= max_messages:
		message = None
		while message is None:
			message = buffer.get_message(time_out)
		blf_writer.on_message_received(message)
		messages +=1
	blf_writer.stop()'
----------------------------------------------------------------------------------------------

GitYom, 2021-05-31:
``

zariiii9003, 2021-06-03:
I fixed your code formatting. Take a look at https://python-can.readthedocs.io/en/develop/listeners.html#can.SizedRotatingLogger .
It might be what you are looking for.

GitYom, 2021-06-04:
Hello Zari,
thanks for the formatting of the code. I used the code balise but it had not the visual effect I hope.  for the next posts, I will now how to do it.

I just had a look to the documention of the development version and the SizedRotatingLogger  seems to do what I expect. I will try it.

Thanks,

Guillaume

felixdivo, 2021-06-14:
@GitYom Great to hear that it helped. If you find issues with that approach (e.g. due to a bug), feel free to open a new issue.

closed 2021-06-14 by felixdivo