hardbyte/python-can #1064: if use default name app_name="CANalyzer", open VN1630A CAN/LIN Interface will fail

xinkunlong, 2021-06-04:
**Describe the bug**

**The vn1630A I am using now has only two valid channels**, so when using VectorBus to only pass in the parameters channel and bitrate, the device cannot be opened because it cannot find a valid hardware type. If the parameter app_name='', then the device can be driven.

**To Reproduce**
VectorBus(channel=0,bitrate=500000)

**Expected behavior*
hw_type, hw_index, hw_channel = self.get_application_config(app_name, channel)
hw_type will equel 0

**Additional context**
If the parameter app_name=''
then the device can be driven.

OS-version: WIN10
Python version:Python 3.8.9 (tags/v3.8.9:a743f81, Apr  2 2021, 11:10:41) [MSC v.1928 64 bit (AMD64)] on win32
python-can version: 4.0.0.dev0
python-can interface/s:  VN1630A CAN/LIN Interface 

  only two valid channels
<details><summary>Traceback and logs</summary>

  

  

```python
def func():
    return 'hello, world!'
```
</details>

zariiii9003, 2021-06-04:
Did you assign the app_name to the channel in Vector Hardware Config?

xinkunlong, 2021-06-04:
No, I did not.   
Pseudocode:
    if  call VectorBus(channel=0,bitrate=500000) 
    hw_type will equel 0, 
    then idx = -1,
    so the device can not  be driven.

    if call VectorBus(channel=0,bitrate=500000,**app_name=''**),
    the device can be driven.

zariiii9003, 2021-06-04:
> No, I did not.

Well you have to.

xinkunlong, 2021-06-08:
yes, thanks.

felixdivo, 2021-06-08:
Great!

closed 2021-06-08 by felixdivo