hardbyte/python-can #1065: pyhton-can 4.0 cannot initialize PCAN-USB

tiger86023086, 2021-06-05:
My code is:
   import can
    mybus = can.interface.Bus(bustype="pcan",channel="PCAN_USBBUS1",bitrate=500000).
When I run it ,the error report:

     File "D:\Python37\lib\site-packages\can\interface.py", line 114, in __new__
    return cls(channel, *args, **kwargs)
  File "D:\Python37\lib\site-packages\can\interfaces\pcan\pcan.py", line 289, in __init__
    raise PcanError(self._get_formatted_error(result))
can.interfaces.pcan.pcan.PcanError: An invalid parameter was specified

I find in pcan.py,the code,the line 243-245:

       hwtype = PCAN_TYPE_ISA
        ioport = 0x02A0
        interrupt = 11
Is there an error or not??

tiger86023086, 2021-06-06:
May be I have solve this bug. 
I cannot find the parameter ' PCAN_ALLOW_ERROR_FRAMES'  in my attechment 'PCANBasi.py' which give when I buy PCAN.
So I think  my PCAN_USB fireware is old ,it cannot have the option of 'PCAN_ALLOW_ERROR_FRAMES'.
When I comment the code:
   ##        result = self.m_objPCANBasic.SetValue(
##            self.m_PcanHandle, PCAN_ALLOW_ERROR_FRAMES, PCAN_PARAMETER_ON
##        )
##
##        if result != PCAN_ERROR_OK:
##            raise PcanError(self._get_formatted_error(result))

It don't appear the error.
But Idon't know my doing have other affect,beacase i cannot get the session code.Please tell the affect of the session code.

davidva-ecco, 2021-06-25:
@tiger86023086 I was unable to reproduce your issue with Windows 10 and PCAN.  Can you verify you installed the PCAN USB drivers as well as the PCANBasic API?

python-can version
```
(venv) \> python -m pip show python-can
Name: python-can
Version: 4.0.0.dev2
Summary: Controller Area Network interface module for Python
Home-page: https://github.com/hardbyte/python-can
Author: Python CAN contributors
Author-email: None
License: LGPL v3
Location: ...lib\site-packages
Requires: windows-curses, pywin32, wrapt, mypy-extensions
Required-by:
```

Test
```
> python
Python 3.9.5 (tags/v3.9.5:0a7dcbd, May  3 2021, 17:27:52) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import can
>>> mybus = can.interface.Bus(bustype="pcan",channel="PCAN_USBBUS1",bitrate=500000)
>>> 
>>> print(f"{mybus.channel_info}")
PCAN_USBBUS1
```

pkess, 2021-12-19:
This is related to pr #1188. I had the same issue and updating pcan basic package on my pc solved the problem. With the mentioned PR you would get an error message when your pcan library version is to old

felixdivo, 2021-12-20:
As this issue is rather stale anyways, let us close it then.

closed 2021-12-20 by felixdivo