hardbyte/python-can #1066: How to send multiple frames using Socket-can in python using can.message class

sakthikappu, 2021-06-05:
I would like send multiple message (10000 canfd frames with variouse arbitration id )  in single can.send() api call. 
How to construct the multiple frames using can.message class.

In the below example , single msg is formed and send via bus.send(msg)
  msg = can.Message(arbitration_id=0xc0ffee,
                      data=[0, 25, 0, 1, 3, 1, 4, 1],
                      is_extended_id=True)

    try:
        bus.send(msg)

In same way how to form a multiple message and send in a single api can.send() call .
Could anyone support on this ?

felixdivo, 2021-06-05:
[You can only send a single frame at a time with the current API.](https://python-can.readthedocs.io/en/develop/bus.html#can.BusABC.send_periodic) You have to loop over the messages that you generate and `bus.send(...)` them individually. If you want to send stuff periodically, there are other options (e.g. see [here](https://github.com/hardbyte/python-can/blob/develop/examples/cyclic.py)).

closed 2021-06-05 by felixdivo

sakthikappu, 2021-06-06:
Thanks for the timely response .
How to construct multiple frames , we shall use a for loop to construct the message 

Is it possible to store multiple frames having different arbitrationid ,data and send them periodically ?
Ex:
msg[i].arbitrationid[i] = i ;
msg[i].data[i] = data;

shall we send this (msg) using periodic apis ?

In simple words , how to store a bulk  multiple frame messages (having various arbitration id , data frames values ) in an array and to push them in a single periodic send api call

Can you please support me on this ?

felixdivo, 2021-06-06:
Well, a `can.Message` is a rather normal python object. You can send them by looping over the collection or by using periodic APIs. Please look at `examples/` in the repository root.

We can unfortunately not provide support here, please use different media for that. StackOverflow might help.