hardbyte/python-can #1078: Codecov in CI often fails without apparent reason

felixdivo, 2021-06-08:
This was notable in many of the recent pull requests. For example: #1075 essentially only removed comments, so it should have nearly no impact on the coverage. However, this is what the codecov bot reports ([here](https://github.com/hardbyte/python-can/pull/1075#issuecomment-856841414)): 

[image]

Besides being annoying, it also breaks the mergify auto-merge rule. Furthermore, it makes determining whether a PR is good to be merged quite hard.

felixdivo, 2021-06-08:
@zariiii9003 mentioned this issue and a possible solution [here](https://github.com/hardbyte/python-can/issues/951#issuecomment-821872240).

hardbyte, 2022-11-15:
Fixed in #1430

closed 2022-11-15 by hardbyte

felixdivo, 2022-11-15:
Thanks @hardbyte! :partying_face: