hardbyte/python-can #1097: No tasks displayed when trying to print "_periodic_tasks"

catmemo, 2021-06-24:
**Describe the bug**

I tried to send a message with below code:
bus.send_periodic(msg_vehspd, 0.01, duration=None, store_task=True)

When I tried to send another data, I would like to stop this task, so I print _periodic_tasks before I call this code again.
However, [] was displayed.

**To Reproduce**
Steps to reproduce the behavior (please add longer code examples below):
```
from dearpygui.core import *
from dearpygui.simple import *
import threading
import can
import math
from time import sleep

add_additional_font('simsun.ttc', 16, glyph_ranges = 'chinese_full')

def get_values(sender, data):
    
    BCS_VehSpd = get_value("车速")
    VCU_GearLvlDisp = get_value("挡位")
    DCU_RotSpdAct = get_value("转速")
    BMS_BattSocDisp = get_value("电池电量")
    BCM_DriverDoorAjarSt = get_value("驾驶员侧门状态")
    BCM_PsngrDoorAjarSt = get_value("前乘客侧门状态")
    BCM_RRDoorAjarSt = get_value("右后侧门状态")
    BCM_RLDoorAjarSt = get_value("左后侧门状态")
    BCM_BonnetAjarSt = get_value("引擎盖状态")
    BCM_TrunkAjarSt = get_value("后背箱状态")
    TPMS_FLTyrePr = get_value("左前轮胎压力值")
    TPMS_FRTyrePr = get_value("右前轮胎压力值")
    TPMS_RLTyrePr = get_value("左后轮胎压力值")
    TPMS_RRTyrePr = get_value("右后轮胎压力值")
    
    period_can_messages_thread = threading.Thread(target=send_period_can_messages, args=(BCS_VehSpd, VCU_GearLvlDisp, DCU_RotSpdAct, BCM_DriverDoorAjarSt, BCM_PsngrDoorAjarSt, BCM_RRDoorAjarSt, BCM_RLDoorAjarSt, BCM_BonnetAjarSt, BCM_TrunkAjarSt, TPMS_FLTyrePr, TPMS_FRTyrePr, TPMS_RLTyrePr, TPMS_RRTyrePr))
    period_can_messages_thread.start()
    

def send_period_can_messages(BCS_VehSpd, VCU_GearLvlDisp, DCU_RotSpdAct, BCM_DriverDoorAjarSt, BCM_PsngrDoorAjarSt, BCM_RRDoorAjarSt, BCM_RLDoorAjarSt, BCM_BonnetAjarSt, BCM_TrunkAjarSt, TPMS_FLTyrePr, TPMS_FRTyrePr, TPMS_RLTyrePr, TPMS_RRTyrePr):
    
    bus = can.interface.Bus(bustype='vector', app_name='CANalyzer', channel=0, bitrate=500000)

##    bus.stop_all_periodic_tasks(remove_tasks=True)
    
    ## 车速

    # 将车速转为hex值，并去掉前两位的0x
    BCS_VehSpd = hex(math.ceil(BCS_VehSpd/0.05625))[2:]
    #  zfill() 方法返回指定长度的字符串，原字符串右对齐，前面填充0
    BCS_VehSpd = str(BCS_VehSpd).zfill(4)
    BCS_VehSpd = str(int(BCS_VehSpd[0]) + 2) + BCS_VehSpd[1:]
    # 构建车速data
    msg_vehspd_data = bytearray([0, 0, 0, 0, int(BCS_VehSpd[:2], 16), int(BCS_VehSpd[-2:], 16), 0, 0])
    log_debug(msg_vehspd_data)
    msg_vehspd = can.Message(arbitration_id=0x260, data=msg_vehspd_data, is_extended_id=False)
    
    Send_BCS_VehSpd = bus.send_periodic(msg_vehspd, 0.01, duration=None, store_task=True)
    
    

show_logger() # were going to use the logger here to show the result

with window("基本数据", width=200, height=200, x_pos=20, y_pos=20):

    add_input_float("车速", callback=get_values, on_enter=True)
    add_combo("挡位", items=["invalid", "D", "N", "R", "P", "Reserved", "Reserved"], callback=get_values)
    add_input_float("转速", callback=get_values)
    add_input_float("电池电量", callback=get_values)

with window("胎压", width=360, height=200, x_pos=230, y_pos=20):

    add_input_float("左前轮胎压力值", callback=get_values)
    add_input_float("右前轮胎压力值", callback=get_values)
    add_input_float("左后轮胎压力值", callback=get_values)
    add_input_float("右后轮胎压力值", callback=get_values)
    
    
with window("车门开启指示灯", width=200, height=200, x_pos=600, y_pos=20):

    add_checkbox("驾驶员侧门状态", default_value=False, callback=get_values)
    add_checkbox("前乘客侧门状态", default_value=False, callback=get_values)
    add_checkbox("右后侧门状态", default_value=False, callback=get_values)
    add_checkbox("左后侧门状态", default_value=False, callback=get_values)
    add_checkbox("引擎盖状态", default_value=False, callback=get_values)
    add_checkbox("后备箱状态", default_value=False, callback=get_values)

set_main_window_size(850, 600)    
start_dearpygui()
```

**Expected behavior**
Stop task "Send_BCS_VehSpd " before sending another data

**Additional context**

OS-version: Win10
Python version: 3.7.2
python-can version: 3.3.4
python-can interface/s: _periodic_tasks

  
<details><summary>Traceback and logs</summary>

  

  

```python
Refer to To Reproduce
```
</details>

felixdivo, 2021-06-24:
@catmemo Could you please remove any code that is not related to the problem, like the GUI setup? That would make finding the issue a lot easier.

catmemo, 2021-06-25:
@felixdivo Done, thks~

catmemo, 2021-06-25:
@felixdivo Done, thks~

> @catmemo Could you please remove any code that is not related to the problem, like the GUI setup? That would make finding the issue a lot easier.

Done, thks~

felixdivo, 2021-06-29:
Ah, thanks for the cleanup. Now I see the issue: The periodic tasks are bound to the instance of `can.interface.Bus`, but you create a new one in each iteration of the timer. However, you should use the *same* bus instance in all invocations of the timer callback. That should do what you want. And also you should close the bus once you are done using it, e.g. with `bus.shutdown()` at the very end of your script.

felixdivo, 2021-06-29:
@catmemo If there is another issue with the setup where you suggest there is a bug lurking in the library, please feel free to open a new issue.

closed 2021-06-29 by felixdivo

catmemo, 2021-06-29:
> Ah, thanks for the cleanup. Now I see the issue: The periodic tasks are bound to the instance of `can.interface.Bus`, but you create a new one in each iteration of the timer. However, you should use [...]

Thank you for your quickly reply！

catmemo, 2021-06-29:
> @catmemo If there is another issue with the setup where you suggest there is a bug lurking in the library, please feel free to open a new issue.

Well, I will. Thanks again.