hardbyte/python-can #1101: Timestamp value back to 0 after a few seconds when logging ASC or BLF

thbuch, 2021-07-02:
Hi there,
As mentioned, I found a strange behaviour sometimes when trying to log data from a CAN Bus into a ASC or BLF file using ASCWriter or BLFWriter. 
In some cases, when checking the messages logged, I found timestamps starting again from zero after first 2-3 seconds.

I noticed too that some messages were missing in comparison with logs taken with a Vector CanCase and its own software Canoe. May it be related?

This is my sample code:

import can
import os
import time
import subprocess

filename = r'LogTest.blf'
logger = can.BLFWriter(filename)

with can.interface.Bus(bustype="socketcan", channel="can0", bitrate=500000) as bus:
    try:
        while True:
            msg = bus.recv(1)
            if msg is not None:
                logger.on_message_received(msg)
            else:
                print("Timeout")
                
    except KeyboardInterrupt:
        pass
    logger.stop()

**Additional context**

OS-version: Raspbian in a Raspberry Pi 3B+
Python version: 3.7.3
python-can version: 3.3.4
python-can interface/s: socketcan with a Seeed-Studio 2 Channel CAN-FD HAT

Thanks you!

Mrunmayee01, 2022-10-07:
make sure all messages have  a timestamp, especially the messages "sent" from the application

thbuch, 2023-01-22:
Well, as you can see in my code, I am just reading, not sending.
I tested another interface and no problem. Not sure if the problem is the hardware itself or its driver. We can discard a problem on the python-can library or my code. So, I close the issue.

closed 2023-01-22 by thbuch