hardbyte/python-can #1122: ValueError: 1207997447 is not a valid XL_ChannelCapabilities

ErdosAndris, 2021-08-28:
https://github.com/hardbyte/python-can/blob/1979a75ce505a7d3a29651bd34f7fc111a33d730/can/interfaces/vector/canlib.py#L801

When I call the VectorBus constructor inside canlib.py to create a new Vector CAN channel, I use the following parameters: 
**channel=0, serial=1347** because according to the API doc, in this way, the channel config has been discovered automatically by device serial number (or at least, as I understood), and only on that specific device

The Traceback and the channel configuration with the serial number:

Traceback (most recent call last):
  File "c:\WORK\VSTARTEST\MultiCoreBus_Integration\Tools\executor\executor.py", line 205, in worker
    assert target.execute(), "Execution failed"
  File "c:/WORK/VSTARTEST/MultiCoreBus_Integration/Tools/executor/executables\framework.py", line 27, in execute
    inspect.getmembers(__import__(suite_path))
  File "c:\WORK\VSTARTEST\MultiCoreBus_Integration\Applications\ScGateway\Opponent\suite.py", line 69, in <module>
    can = CanVectorChannel(channel=0, serial=1347)
  File "c:\WORK\VSTARTEST\MultiCoreBus_Integration\Tools\framework\channel\CanVectorChannel.py", line 210, in __init__
    channel_configs = get_channel_configs()
  File "c:\WORK\VSTARTEST\MultiCoreBus_Integration\Tools\framework\channel\CanVectorChannel.py", line 852, in get_channel_configs
    xlcc.channelCapabilities
  File "c:\WORK\xtools\python3.5\lib\enum.py", line 241, in __call__
    return cls.__new__(cls, value)
  File "c:\WORK\xtools\python3.5\lib\enum.py", line 476, in __new__
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 1207997447 is not a valid XL_ChannelCapabilities

[image]

My question: can you please take a look at this behaviour, and propose a quick workaround if possible? Unfortunately, I didn't get enough experience with this code yet to raise a solution or contribute to it

zariiii9003, 2021-08-31:
python-can 4.0 is only supported for python 3.6 or newer. There were some changes in for the enum module between python 3.5 and 3.6.

Please try again with a newer python version and report back.

felixdivo, 2021-09-27:
If the issue persists, feel free to comment again. I'll close it for now.

closed 2021-09-27 by felixdivo