hardbyte/python-can #1130: How Can I work with USB CAN Adapter Robotell Co.?

yosefgouda2020, 2021-09-26:
Hello , 

I bought USB CAN Adapter Robotell Co. from Aliexpress in this [link ](https://www.aliexpress.com/item/32959205216.html?spm=a2g0o.search0302.0.0.413932e3QFAPzt&algo_pvid=04205ea5-f1d8-42fe-b700-0f97463eb722&algo_exp_id=04205ea5-f1d8-42fe-b700-0f97463eb722-2)

I am not able to find software for it is there any method to run this hardware ?

My operating system is windows.

Thanks,

felixdivo, 2021-09-27:
[Here](https://python-can.readthedocs.io/en/develop/interfaces.html) are the interfaces supported by version [`4.0.0-dev.2`](https://github.com/hardbyte/python-can/tree/4.0.0-dev.2). Maybe that helps (you can try a few).

In general, we should consider providing more beginner-friendly ways to find the right interface.

yosefgouda2020, 2021-09-27:
Hi @felixdivo Thank you for your help 
I tried to use the python script of this [link ](https://github.com/hardbyte/python-can/blob/4.0.0-dev.2/can/interfaces/robotell.py) but it didn't work when I run it using python so is there any steps to make USB CAN Adapter Robotell working and display the CAN messages.
Thank you.

felixdivo, 2021-09-27:
How did you use that "script"? You should use it more like this: https://github.com/hardbyte/python-can/blob/develop/examples/receive_all.py.

felixdivo, 2021-09-28:
Okay, so maybe [this](https://realpython.com/lessons/scripts-modules-packages-and-libraries/) helps with the terminology. Generally, we can't give general programming advice here, there are other platforms for that.

You should copy the script I linked above or other files from [the example forlder](https://github.com/hardbyte/python-can/blob/develop/examples) and modify the part where the `can.interface.Bus` instance is created. In particular, in the linked file you have to specify `bustype="pcan", channel="PCAN_USBBUS1", bitrate=250000` in line 15. You can set different `bustype`s and that will cause the right interface of [these](https://python-can.readthedocs.io/en/develop/interfaces.html) to be selected.

If you have a more specific issue, please feel free to comment. If you think that more documentation is needed, feel free to suggest some.

felixdivo, 2021-09-28:
I opened #1131 for the general discussion and will close this one for now. If you find a **bug**, please comment back.

closed 2021-09-28 by felixdivo

yosefgouda2020, 2021-09-29:
@felixdivo I changed the code in the receive_all.py line 14 like this

    with can.interface.Bus(
        bustype="robotell", channel="COM3", ttyBaudrate=115200,bitrate=250000
    ) as bus:

and I got this error 

 import can
  File "C:\Python39\lib\site-packages\can\__init__.py", line 17, in <module>
    from .listener import Listener, BufferedReader, RedirectReader, AsyncBufferedReader
  File "C:\Python39\lib\site-packages\can\listener.py", line 9, in <module>
    from can.message import Message
  File "C:\Python39\lib\site-packages\can\message.py", line 11, in <module>
    from . import typechecking
  File "C:\Python39\lib\site-packages\can\typechecking.py", line 9, in <module>
    import typing_extensions
ModuleNotFoundError: No module named 'typing_extensions'

Also i added the CAN module in this path C:\Python39\lib\site-packages\can\ 
and if there is an example run on robotell it will be great.

Thank you.

felixdivo, 2021-09-29:
The code looks good on first glance. Did you install the library with pip?

yosefgouda2020, 2021-09-30:
yes using pip **pip install python-can**, and also if there are some steps to setup and run for robotell hardware you know and could you please share with me?

felixdivo, 2021-09-30:
Sorry, I can't tell you more than what's here: https://python-can.readthedocs.io/en/develop/interfaces/robotell.html

zeroboy0010, 2022-12-01:
import can 
from can.interfaces import robotell
import time

bus = robotell.robotellBus(ttyBaudrate=115200, channel='/dev/ttyACM0', bitrate=None)
msg = can.Message(arbitration_id=0x101,
                  data=[100, 100, 100, 100, 100, 100, 100, 100],
                  is_extended_id=False)
while(1):
    # s
    msg1 = bus.recv(1)
    msg2 = bus.recv(1)
    print(msg1)
    print(msg2)