hardbyte/python-can #1133: new release

danielhrisca, 2021-09-29:
Hello all, is there some target release date for version 4.0.0 final?

felixdivo, 2021-11-12:
Now, #1046 is "only" waiting for reviews of the changes and merges. The implementation is done.

felixdivo, 2021-12-14:
Yay, #1046 is complete. I'll provide a release candidate soon (when time permits).

felixdivo, 2022-01-16:
#1216 adds a changelog

felixdivo, 2022-01-20:
Waits for a regression to be fixed by #1221. Then there will be version `4.0.0-rc.0` and an announcement issue.

felixdivo, 2022-01-28:
Okay, so the blocking #1221 and #1230 are resolved. Everything is implemented for the release.

felixdivo, 2022-02-04:
A new release candidate is available, [`4.0.0rc0`](https://github.com/hardbyte/python-can/releases/tag/4.0.0rc0)! :partying_face:

You can install it with `pip install python-can==4.0.0rc0`.

felixdivo, 2022-02-04:
Btw, #1249 is not included in that release. It will be in the next one or the full release, depending of whether we need a second *rc* build at all.

felixdivo, 2022-02-04:
See  #1250. Thank you for your patience. :smiley:

closed 2022-02-04 by felixdivo