hardbyte/python-can #1134: On transmitting, when vector CAN channel is not connected, exception or error is not observed.

81piyush, 2021-09-29:
Hello, I am using the python-can module on Windows platform with Vector Hw VN1630A connected. 

On disconnecting the CAN channel 1 of this Hw (not connected to any of the ECU node or CAN receiver, so that acknowledgment is not received).

Then on attempt to transmitting the CAN data through this channel using the "send" or "send_periodic" methods, even though the data is not through, there is no error or exception by the these methods.

On using the following code:

import can

try:
    bus = can.Bus( bustype='vector', channel=0, bitrate=500000 )
except Exception as e:
    print("Could not initialize the Vector with exception: ", e)

canmsgtx = can.Message(arbitration_id=0x123, is_extended_id=True, data=[0x11, 0x22, 0x33])

try:
    bus.send(canmsgtx, timeout=0.2)
except Exception as e:
    print("Could not transmit the data using the Vector with exception: ", e)
    if e==can.CanError:
        print("Vector HW is not online!")

print("Vector Bus status is : ", bus.state)

I receive the following response:
Vector Bus status is :  BusState.ACTIVE

christiansandberg, 2021-09-30:
A send operation only adds the message to a buffer on the hardware or driver. As long as there is room for more messages in the queue the call will be successful. It is only until the buffer becomes full when it will raise an exception.

81piyush, 2021-10-05:
Thanks. I have tried sending the message in the loop for over 1500 iterations. At around 1215, the buffer was full and the following exception was raised.
xlCanTransmit failed (XL_ERR_QUEUE_IS_FULL)

closed 2021-10-05 by 81piyush