hardbyte/python-can #1161: can.CanError not being raised

menroka, 2021-11-02:
I have an Ixxat USB-to-CAN V2 hooked up to an embedded system.
The system is initially shut down, without any electricity, so there is no way for the CAN to work, because there is only one lonely Ixxat and nothing can send the acknowledgement bit.
When calling network.send_message(canID, data) like this:
```
while tryingToEnterBackdoor:
    try:
        network.send_message(secretID, backDoorData)
        tryingToEnterBackdoor = False
    except can.CanError:    # this error is currently not working as intended
        time.sleep(0.002)
```
The Ixxat then starts flashing the red CAN LED and the console is outputting "CAN acknowledgment error" in red letters. So far this is expected.
However, the function network.send_message does not raise the can.CanError exception and instead we leave the loop instantly.

felixdivo, 2021-11-10:
Hey @menroka, what piece of code are you referencing? Is it on the `develop` branch?

menroka, 2021-11-10:
Hi @felixdivo , the referenced code block is something I have written myself using the [canopen](https://canopen.readthedocs.io/en/latest/) packet.
The send_message function is documented [here](https://canopen.readthedocs.io/en/latest/network.html#canopen.Network.send_message).