hardbyte/python-can #1169: ASCReader can support both dec and hex base

tomguo2000, 2021-11-15:
**Is your feature request related to a problem? Please describe.**
with can.ASCReader ('2021-09-09_07-25-17.Part002.asc') as can_log:
    for msg in can_log:
        print(msg.arbitration_id, msg.data)

it will appear : 
frame.append(int(byte, 16))
ValueError: byte must be in range(0, 256)

**Describe the solution you'd like**
add more parameter when load asc file

**Describe alternatives you've considered**
None

**Additional context**
some can message record such as GL-3000, will output asc such as:

date Thu Sep 2 09:37:49 am 2021
base dec timestamps absolute
Begin Triggerblock Thu Sep 2 09:37:49 am 2021

770.592543 7  1282       Rx D 8   0   0   0   0   0 170   0   0
770.592543 9  800        Rx D 8   0  64   0   0  75   0   0   0
...

pierreluctg, 2021-11-19:
This is already supported by the `ASCReader` in develop branch, not yet in master.

See: https://github.com/hardbyte/python-can/blob/develop/can/io/asc.py#L73-L81

felixdivo, 2021-12-22:
I assume this is resolved. If not, please comment.

closed 2021-12-22 by felixdivo