hardbyte/python-can #1171: Add a method to the message-class that gives a bit-representation of a message

MarcusVH98, 2021-11-15:
**Is your feature request related to a problem? Please describe.**
I need to construct raw bit representations of can-frames.
Context: I am working on a project where I need to communicate wirelessly with a CAN-bus to receive data, and send a couple commands. I am planning to do this with a simple TCP-interface, where I send the can-frame to a node on the bus, which takes it out of the TCP-packet and distributes it on the bus. Currently I haven't found a way to use the message-class without a bus-interface, but I don't think such an interface would work in my case, as I can't physically connect to the bus in any way.

**Describe the solution you'd like**
A method in the message-class that outputs a byte-stream, or similar, of how the message would look on a real can-bus.

**Describe alternatives you've considered**
I considered constructing such a representation myself, from the fields in the message class, but I am not confident I can reconstruct such a frame with every field in the correct order, and with fields for error-detection etc. without creating bugs/errors.

**Additional context**
The project runs on PySide6. I am still new to making issues, so if something is wage or whatever, please let me know.

felixdivo, 2021-12-07:
Did you consider running `python-can` on the other node for hardware interfacing and using some other TCP-based interface that `python-can` supports for the transport to the other node?

MarcusVH98, 2022-01-16:
Thank you for the suggestion. The embedded micro-controllers can only run C, so I'm not sure that would work. After digging some more in the source code I found in `can.interfaces.socketcan.socketcan` the functions `build_can_frame` and `dissect_can_frame` which seem to do what I want though. I will test it out on the hardware in a couple weeks.

felixdivo, 2022-01-16:
Keep in mind that the functions in `can.interfaces.socketcan` translate between python-can and the socketcan driver of Linux. Those are not the bit representations that you see on the wire!

MarcusVH98, 2022-01-17:
I see, that would be a problem for me. I was thinking, since python-can supports interfacing with a physical can bus, there is a conversion to a "real" can frame that would be observed on the wire, is that not the case? Is that conversion in the different drivers utilized? I tried finding such a conversion in socketcan, but was unable to. I'm considering giving it a go to manually encode/decode.

felixdivo, 2022-01-17:
I don't know of any driver that does this. It's usually done by a more low-level component. (Note: You would also have to consider bit stuffing.)

MarcusVH98, 2022-01-18:
Thanks for the reminder, I'll make sure to account for bit stuffing. And thank you so much for the help, I'm sure I'll figure something out. :-)

closed 2022-01-18 by MarcusVH98

felixdivo, 2022-01-18:
Happy to help. :)

You might want to look at [`bitstring`](https://bitstring.readthedocs.io/en/latest/packing.html), it might be trivial to implement (all but bit stuffing) in there.