hardbyte/python-can #1176: 10ms periodic CAN frame exported in 17 ms

leu1dy, 2021-11-29:
Hello,

For automotive purpose, i need to export CAN frames every 10ms via PCAN device.

I made a small script for test to export this CAN frame but period is wrongly managed.

For 10ms, pcan view tool is showing a period of 17 ms.
For 20ms, pcan view tool is showing a period of 27 ms.

Here is the script :
done = False
bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000)     
msg = can.Message(arbitration_id=0x432, data=[1, 2, 3, 4, 5, 6], is_extended_id=False)       
task = bus.send_periodic(msg, 0.01)
assert isinstance(task, can.CyclicSendTaskABC)

while not done:
    test=1

OS-version:
Window 10
Python version: 3.7
python-can version: 3.3.4

Why is there such drift in the CAN frame period ? Something wrong in my script ?
Thank you.

christiansandberg, 2021-11-30:
Could you try using develop? There have been some improvements to periodic transmissions since last release.

leu1dy, 2021-11-30:
Thank you so much, it's more reliable with the develop branch.