hardbyte/python-can #1177: ics will crash when FD message's payload size is over 64 bytes

jamesxin, 2021-11-30:
**Describe the bug**
A clear and concise description of what the bug is.
If we send a FD message which is larger than 64 bytes ICS will crash instead of raise an exception. 
**To Reproduce**
Steps to reproduce the behavior (please add longer code examples below):
     ign_data = [1]
     ign_data.extend([0]*64)

        frame = can.Message(
            arbitration_id=0x3B3,
            data=ign_data,
            is_fd=self.is_fd,
            is_extended_id=False
        )
        bus.send_periodic(frame, period,  timeout)

**Expected behavior**
A clear and concise description of what you expected to happen.

Throw an exception for a FD message with data over limit.

**Additional context**

OS-version:
Python version:
python-can version:
python-can interface/s:

  
<details><summary>Traceback and logs</summary>

  

  

```python
def func():
    return 'hello, world!'
```
</details>

pierreluctg, 2021-11-30:
Hi @jamesxin can you please the version of `python-ics` that you are using and the stack trace if you are getting one?

jamesxin, 2021-12-01:
I am using python-ics                        905.11.post1.

     Since this is a crash there is no exception for stack traces. I
collect the stack trace before we crash as below.

Traceback (most recent call last):
  File ".\FD_over_64_tester.py", line 53, in <module>
    send()

You can use the following code to reproduce the issue.

import time
import traceback, sys
import can
config = {'bustype': 'neovi', 'serial': 'V23137', 'bitrate': 2000000,
          'channel': 1, 'data_bitrate': 2000000, 'fd': True}
bus = can.interface.Bus(**config)
bus.set_filters([{"can_id": 0x7E0, "can_mask": 0xFFFF, "extended": False},
                 {"can_id": 0x7E8, "can_mask": 0xFFFF, "extended": False},
                 {"can_id": 0x3B3, "can_mask": 0xFFFF, "extended": False},
                 ])

def send():
    data = [0x02, 0x10, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, ]
    data.extend([0]*64)
    msg = can.Message(arbitration_id=0x7E8,
                      data=[0x02, 0x10, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00,
],
                      extended_id=False,
                      is_fd=True)
    stack = traceback.extract_stack()
    trc = 'Traceback (most recent call last):\n'
    stack_str = trc + ''.join(traceback.format_list(stack))
    print(stack_str)
    try:
        bus.send(msg)
        print("Message sent on {}".format(bus.channel_info))
    except can.CanError:
        print("Message NOT sent")

if __name__ == '__main__':
    send()

Bests,

James

pierreluctg, 2021-12-02:
@jamesxin please also note that `2000000` is not a valid `bitrate` value. In CAN FD this bitrate value is used for the arbitration phase.
You probably want the values to be set to `bitrate=500000` (arbitration phase) and `data_bitrate=2000000` (data phase).

jamesxin, 2021-12-02:
@pierreluctg,
       Thank you for pointing that out. I am currently using 3.9.6.7.
Unfortunately It still crashed after I updated the bitrate to 500K. As I
remember, 'bitrate' should be ignored for CANFD.

Bests,

James

pierreluctg, 2021-12-02:
@jamesxin the bitrate is still use in the case of CAN FD and this bitrate will be use for the data phase and the data phase if bit rate switching is not enabled. If bit rate switching is enabled then the data bitrate will be use for the data phase only.

In order to get more information on the issue you are experiencing, can you please enable the Python fault handler when running your code?

You can also enable the fault handlers at startup by setting the PYTHONFAULTHANDLER environment variable or by using the -X faulthandler command line option.

https://docs.python.org/3/library/faulthandler.html#module-faulthandler

jamesxin, 2021-12-02:
Yes. Then we have the same understanding for the usage of bitrate.

I tried '-X faulthandler' but unfortunately it didn't work.

 *python -X faulthandler .\FD_over_64_tester.py*
*The can_config used {'bustype': 'neovi', 'serial': 'V23137', 'bitrate':
500000, 'channel': 1, 'data_bitrate': 2000000, 'fd': True}*
*Traceback (most recent call last):*
*  File ".\FD_over_64_tester.py", line 33, in <module>*
*    send()*

*Message sent on ValueCAN 4-2 V23137 CH:[1]*

'-m trace' actually shows some interesting results. The failure actually
happened on the print line. But if we remove the 'bus.send(msg)',
everything is fine.
It looks like somehow the bus.channel_info contains some invalid data after
bus.send.

*FD_over_64_tester.py(29):     bus.send(msg)*
* --- modulename: neovi_bus, funcname: send*
*neovi_bus.py(359):         if not ics.validate_hobject(self.dev
<http://self.dev>):*
*neovi_bus.py(361):         message = ics.SpyMessage()*
*neovi_bus.py(363):         flag0 = 0*
*neovi_bus.py(364):         if msg.is_extended_id:*
*neovi_bus.py(366):         if msg.is_remote_frame:*
*neovi_bus.py(369):         flag3 = 0*
*neovi_bus.py(370):         if msg.is_fd:*
*neovi_bus.py(371):             message.Protocol = ics.SPY_PROTOCOL_CANFD*
*neovi_bus.py(372):             if msg.bitrate_switch:*
*neovi_bus.py(374):             if msg.error_state_indicator:*
*neovi_bus.py(377):         message.ArbIDOrHeader = msg.arbitration_id*
*neovi_bus.py(378):         msg_data = msg.data*
*neovi_bus.py(379):         message.NumberBytesData = len(msg_data)*
*neovi_bus.py(380):         message.Data = tuple(msg_data[:8])*
*neovi_bus.py(381):         if msg.is_fd and len(msg_data) > 8:*
*neovi_bus.py(382):             message.ExtraDataPtrEnabled = 1*
*neovi_bus.py(383):             message.ExtraDataPtr = tuple(msg_data)*
*neovi_bus.py(384):         message.StatusBitField = flag0*
*neovi_bus.py(385):         message.StatusBitField2 = 0*
*neovi_bus.py(386):         message.StatusBitField3 = flag3*
*neovi_bus.py(387):         if msg.channel is not None:*
*neovi_bus.py(389):         elif len(self.channels) == 1:*
*neovi_bus.py(390):             message.NetworkID = self.channels[0]*
*neovi_bus.py(395):         try:*
*neovi_bus.py(396):             ics.transmit_messages(self.dev
<http://self.dev>, message)*
*FD_over_64_tester.py(31):     print(f"Message sent on {bus.channel_info}")*
* --- modulename: cp1252, funcname: encode*
*cp1252.py(19):         return
codecs.charmap_encode(input,self.errors,encoding_table)[0]*
*Message sent on ValueCAN 4-2 V23137 CH:[1] --- modulename: cp1252,
funcname: encode*
*cp1252.py(19):         return
codecs.charmap_encode(input,self.errors,encoding_table)[0]*

Bests,

James

pierreluctg, 2021-12-03:
@jamesxin can you share your `FD_over_64_tester.py` (attach it to your post)?

`bus.channel_info` is a string set in the constructor of the bus and do not change after. From what I understand in the trace log that you shared, this value/string is valid after send and displayed in your trace (four lines from the bottom).

If you are remove `print("Message sent on {}".format(bus.channel_info))` after sending the message do you still see the issue?
Is the frame sent on the actual CAN bus?

jamesxin, 2021-12-03:
`import can
import faulthandler

faulthandler.enable()
config = {'bustype': 'neovi', 'serial': 'V23137', 'bitrate': 500000,
          'channel': 1, 'data_bitrate': 2000000, 'fd': True}
print(f"The can_config used {config}")
bus = can.interface.Bus(**config)
bus.set_filters([{"can_id": 0x7E0, "can_mask": 0xFFFF, "extended": False},
                 {"can_id": 0x7E8, "can_mask": 0xFFFF, "extended": False},
                 {"can_id": 0x3B3, "can_mask": 0xFFFF, "extended": False},
                 ])

print(f"bus channel_info {bus.channel_info}")
def send():
    data = [0x02, 0x10, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, ]
    data.extend([0]*64)
    msg = can.Message(arbitration_id=0x7E0,
                      data=data,
                      extended_id=False,
                      is_fd=True)
    bus.send(msg)
    print(f"Message sent on {bus.channel_info}")

if __name__ == '__main__':
    send()
`

pierreluctg, 2021-12-03:
@jamesxin, ` 0xc0000374` is a heap issue.

Will fix this by avoiding passing maximum of 64 bytes of data to the driver.

Now the question is are you actually trying to send more than 64 bytes of data over a single frame? This will not be possible regardless of this error. A CAN FD frame supports a maximum of 64 bytes of data.

jamesxin, 2021-12-03:
No. We won't send over 64 bytes in one FD frame. This issue is discovered by an accident.

felixdivo, 2021-12-04:
> No. We won't send over 64 bytes in one FD frame. This issue is discovered by an accident.

Still thanks for the report, @jamesxin.

commit 07c8999b (2021-12-03): Truncating message data if over max frame length

Fixes #1177
commit 2275ab2a (2021-12-07): Truncating message data if over max frame length

Fixes #1177
PR #1181 (2021-12-07): Raise ValueError if message data if over max frame length
Fixes #1177
commit 8aa7166c (2021-12-07): Truncating message data if over max frame length (#1181)

Fixes #1177
closed 2021-12-07 by pierreluctg