hardbyte/python-can #1183: Current DLC didn't support FD

jamesxin, 2021-12-08:
### Describe the bug

Technically DLC is only 4 bits long. DLC values ranging from 1001 to 1111 are used to specify the data lengths of 12, 16, 20, 24, 32, 48, and 64 bytes.  Currently Message.DLC is actually the data length instead of DLC. It is ok for classic CAN but for CAN FD it will cause the DLC value up to 64.

Timestamp:    30969.422452        ID: 03b3    S     F BS       DLC: 16    44 00 00 0c e6 00 00 00 00 00 00 00 00 00 00 00

### To Reproduce

### Expected behavior

### Additional context

OS and version:
Python version:
python-can version:
python-can interface/s (if applicable):

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

hartkopp, 2021-12-08:
The DLC is something that happens on the wire.
For SocketCAN we decided to use and provide the 'plain length' value which is intuitive to use by application programmers.

We try to get rid of the can_dlc element in the struct can_frame and maintain it for compatibility reasons inside a union:
https://github.com/linux-can/can-utils/commit/c398e56afb9b60441232736e05fdfb5a525363ca#diff-befd25a84cf5bc1498493804683cc14e493afa2c8ca707f3803c36841347489c

The programmer should never be forced to do DLC <-> Length conversions with that spooky conversion table on his own ...

jamesxin, 2021-12-08:
I don't think python-can having DLC issue on sending FD message. However, in the logging python-can is using data length for DLC. That is a miss leading for the CAN FD message.

zariiii9003, 2021-12-08:
We could replace the `"DLC"` with something like `"LEN"` in the `Message.__str__()` method. I agree with @hartkopp that the length is more useful than the length code.

jamesxin, 2021-12-10:
I would say "DL" (data length) is a good option.

hartkopp, 2021-12-11:
> I would say "DL" (data length) is a good option.

Yes. I think "DL" as "data length" is also good in opposition to the  "DLC" as "data length _code_".

That makes it very clear that "DL" has no code ;-)

PR #1212 (2022-01-23): Change DLC to DL in Message.__str__()
Closes #1183
closed 2022-01-23 by felixdivo