hardbyte/python-can #1186: BLFParseError when handling Dante test suite files

catosh, 2021-12-09:
### Describe the bug
When importing a blf generated with Dante toolsuite, files containing LOBJ with obj_type = 128d shall not set next_pos += obj_size % 4.

### To Reproduce
I can't provide the full blf file, but this chunk leads to the error: 
This part of the LOBJ has the obj size set to 4B
4C 4F 42 4A 20 00 01 00 4B 00 00 00 80 00 00 00 02 00 00 00 0C 00 00 00 25 92 18 68 4F 00 00 00 01 00 00 00 12 00 00 00 0D 00 00 00 4C 6F 67 5F 4B 6F 6D 70 6C 65 74 74 65 72 5F 42 55 53 44 69 72 65 63 74 4C 6F 67 67 69 6E 67 4C 4F 42 4A 20 00 01 00 8F 00 00 00

### Expected behavior

### Additional context
tested on Windows10 64 bit, python 3.7.
Name: python-can
Version: 3.3.4

To fix, I added in blf.py at line 205 following lines:
                    if obj_type != CAN_FD_MESSAGE_64:
                        if obj_type == 128:
                            pass
                        else:
                            next_pos += obj_size % 4
</details>

christiansandberg, 2021-12-10:
Have you tried with the develop branch? It has been changed to seek for the next object instead of calculating its offset.

catosh, 2021-12-14:
Hi christian, 
thanks for the hint. I'll give a try and provide some feedback.

felixdivo, 2022-01-15:
@catosh How did it go?

catosh, 2022-01-16:
Had not the chance yet to check. I plan to test during WW3.

felixdivo, 2022-02-06:
Any updates on this, @catosh? :smiley:

catosh, 2022-02-08:
Hello,
looks like it is working (just removed can-fd IDs and my user from the
screenshots):

import can

if __name__ == '__main__':
    print(can.__version__)
    data = can.BLFReader( "log2.blf")
    for elem in data:
        print(elem)

4.0.0-dev.2

Timestamp: 1634296999.189272    ID: X Rx     F BS       DLC:  8    02
3e 80 55 55 55 55 55     Channel: 0
Timestamp: 1634297010.304660    ID: X Rx     F BS       DLC:  8    02
3e 80 55 55 55 55 55     Channel: 0

...

Timestamp: 1634297017.023856    ID: X Rx     F BS       DLC: 64    12
74 36 01 00 00 00 20 d5 00 ff 1f 11 01 ff 1f 17 01 ff 1f 95 01 ff 1f
8c 4a 13 78 13 f0 70 0f 1c bf 4f f0 70 0c 82 f8 00 c0 d0 71 88 6a 00
0c 90 71 88 6a 00 0a 50 71 88 6a 10 71 c8 6a b0 b1    Channel: 0

vs

C:\Users\\Documents\Coding\pythonProject\pythoncanmain\venv\Scripts\python.exe
C:/Users//Documents/Coding/pythonProject/pythoncanmain/main.py

3.3.4
<can.io.blf.BLFReader object at 0x000002D60E0332B0>
Traceback (most recent call last):
  File "C:/Users//Documents/Coding/pythonProject/pythoncanmain/main.py",
line 7, in <module>
    for elem in data:
  File "C:\Users\\Documents\Coding\pythonProject\pythoncanmain\venv\lib\site-packages\can\io\blf.py",
line 195, in __iter__
    raise BLFParseError()
can.io.blf.BLFParseError

Process finished with exit code 1

felixdivo, 2022-02-09:
Great! (There is an even newer version now, btw. 😄  #1250)

I'll close this then. Thanks for having a second look.

closed 2022-02-09 by felixdivo

catosh, 2022-02-09:
ok, I'll test also with that then :)