hardbyte/python-can #1190: Assigning Can Filters for mcp2515 CAN bus module on python

MohamedJP, 2021-12-13:
so I am using the following method to filter messages between 2 controllers,but the issue with this method is that I can only filter up to 5     
can_ids, so I was wondering if there is a better method to do that, knowing I am quite new to this and to python even.
bus.set_filters([{"can_id": 0x591, "can_mask": 0xFFFFFFF, "extended": False},\
    {"can_id": 0x592"can_mask": 0xFFFFFFF, "extended": False},\
    {"can_id": 0x593, "can_mask": 0xFFFFFFF, "extended": False},\
    {"can_id": 0x594, "can_mask": 0xFFFFFFF, "extended": False},\
    {"can_id": 0x595, "can_mask": 0xFFFFFFF, "extended": False}])

felixdivo, 2021-12-14:
Why are you limited to 5 filters? And on which hardware/library version?

MohamedJP, 2022-01-13:
So sorry for the late reply, and thanks for your prompt reply @felixdivo 
I am building can bus for transmission between multiple rasberry pies, and I wanna use the bus filter to limit the data processed at each node, but this way can only limit up to 5 signals ( 5 can IDs) if I added anymore the filter will not work correctly for some reason.

felixdivo, 2022-01-13:
Interesting. So do you use socketcan? That's the interface "type" (the driver, if you want) that you set in `can.Bus(interface='socketcan', ...)`.

MohamedJP, 2022-01-13:
@felixdivo Yes，I do use socketcan as my interface with can0 as the channel, the whole program works just fine, only this issue is happening, but I wonder if @zariiii9003  solution did fix the problem or  not, I actually dont get it, did he update the whole library (python-can or can-utils ) ? 

how to utilize (Add utility function create_filter) ?

zariiii9003, 2022-01-16:
> @felixdivo Yes，I do use socketcan as my interface with can0 as the channel, the whole program works just fine, only this issue is happening, but I wonder if @zariiii9003 solution did fix the problem [...]

It is just a pull request for now, it is not part of the python-can codebase (and maybe it won't be). The pull request contains a convenience function to calculate the filter.

Regarding your question: You only need one `can_id` and one `mask`:
```
b10110010001 | id:   0x591
b10110010010 | id:   0x592
b10110010011 | id:   0x593
b10110010100 | id:   0x594
b10110010101 | id:   0x595
-------------------------
b10110010000 | code: 0x590
b11111111000 | mask: 0x7f8
```

The `mask` defines which bits are relevant and the `can_id`/`code` defines the value of these bits.