hardbyte/python-can #1203: is there support for mcp2515 spi interfaced with Jetson Nano

prajwal-mulkalwar, 2021-12-30:
Im working on to intergrate  Jetson nano  through J1939 CAN, Im planning to use mcp2515 for can interface with nano, do have support for this interface

felixdivo, 2022-01-15:
It should work with [the `socketcan` interface](https://python-can.readthedocs.io/en/develop/interfaces/socketcan.html) according to [this](https://www.beyondlogic.org/adding-can-controller-area-network-to-the-raspberry-pi/):

> The Linux kernel supports CAN and includes SocketCAN drivers for the Microchip MCP2515 Stand-alone CAN Controller with SPI Interface.

If you have problems, fell free to comment on this/reopen this.

felixdivo, 2022-01-15:
Apparently the chip is supported:  #1190

closed 2022-01-15 by felixdivo

prajwal-mulkalwar, 2022-01-24:
Hi Thank you for response.
how do implement J1939 protocol with this, can you provide any guidelines or examples

felixdivo, 2022-01-24:
Please get it to run on socketcan first. You can check whether it works with `can_utils`. Then run the example from the `README.md`. You also need to set the `channel=...` to the correct interface, you can list all interfaces by `ip a` (it's probably `channel="can0"`).