hardbyte/python-can #1205: Types from typechecking.py should be exported for plugin interfaces

John-Titor, 2022-01-03:
### Describe the bug
An interface plugin cannot fully implement typechecking due to not having access to some internal types; at least AutoDetectedConfig, Channel, and CanFilters could be exported in __init__.py to make this possible.

python-can version: HEAD

zariiii9003, 2022-01-05:
```python3
from can.typechecking import AutoDetectedConfig, Channel, CanFilters
```
This doesn't work for you? Are you using the develop branch?

John-Titor, 2022-01-05:
Well, that's embarrassing. It certainly didn't when I tried earlier, but in attempting to reproduce in order to respond it's clearly working now, so I must have been doing something else wrong. Sorry about that!

closed 2022-01-05 by John-Titor