hardbyte/python-can #1211: Vector hardware,no ack can not raise can.CanError

XXIN0, 2022-01-14:
### Describe the bug

HI,
I want to test the can transmit without ack,but it can raise can.CanError.
[image]

### Additional context

Python version:3.7.9
python-can version: 3.3.4
python-can interface/s (if applicable): Vector CANcaseXL

```python
import can

my_bus = can.interface.Bus(bustype='vector', channel='0,1', app_name='Python Can',bitrate=500000, receive_own_messages=False)
msg = can.Message(arbitration_id=0x123, data=[1, 2, 3, 4, 5, 6, 7, 8], channel="0,1",)
my_bus.send(msg, timeout=0.5)

```

zariiii9003, 2022-01-15:
You need a another device on the physical CAN bus to acknowledge your messages. Otherwise you have the option to use the Virtual Channel in Vector HW config. Then you don't need any receivers.

Would you mind posting your Traceback?

XXIN0, 2022-01-17:
Hi，
It didn't have Traceback,if use my way to send message.I think when the bus timeout,it will raise an error.

If I use the software:"Busmaster",it will raise an error,I don‘t know the diffrence between the "python-can" and "busmaster"
[image]
[image]

felixdivo, 2022-01-17:
Sorry @XXIN0, but I do not understand how python-can comes into play here where the problem is that you want to report or get assistance with.

XXIN0, 2022-01-17:
for example，if I don't connect any devices,when I send an can message,it will raise an error that tell me the send is not ok.

zariiii9003, 2022-01-17:
Which one of these options is your problem?

1. You expect an error, but you don't get any.
2. You excpect no error, but you get can.CanError.

XXIN0, 2022-01-17:
1 is my expect

geynis, 2022-06-02:
Hi, having the same issue, I see there is no implementation in the python canlib for raising errors when sending can frames with Vector interface. Will this be resolved soon?

geynis, 2022-06-07:
@XXIN0 I have found a work around, open the bus with receive_own_messages=True and use the receive function to get an error frame for an error on transmitted frames.
I'm checking Vector's API to see if how can we fix this issue, will let you know.

zariiii9003, 2022-06-07:
for CAN you can check `Message.is_error_frame`. I don't think there is a way to get the exact kind of error.
For CAN FD you can get the error type [like this](https://github.com/hardbyte/python-can/issues/909#issuecomment-703072365).

geynis, 2022-06-07:
@zariiii9003 thanks, i didn't look on the latest version. I still think there should be a better way to get the Tx error on send function and not throw the receive function.

XXIN0, 2022-06-08:
> @XXIN0 I have found a work around, open the bus with receive_own_messages=True and use the receive function to get an error frame for an error on transmitted frames. I'm checking Vector's API to see how can we fix this issue, will let you know.

@geynis Thanks,I'll try your  suggestion.

closed 2023-02-03 by XXIN0