hardbyte/python-can #1219: Is Travis CI disabled?

zariiii9003, 2022-01-16:
[The last Travis CI run was 7 months ago](https://app.travis-ci.com/github/hardbyte/python-can/builds). 
We still have our socketcan tests and mypy on Travis only. Also the Travis badge link is broken.

felixdivo, 2022-01-16:
Well, the `.travis.yml` file is still in the repo.

zariiii9003, 2022-01-16:
Maybe @hardbyte needs to grant Travis access rights again

zariiii9003, 2022-03-02:
@hardbyte can you fix this?

zariiii9003, 2022-08-12:
@hardbyte a little reminder: the socketcan tests are still not run

hardbyte, 2022-11-16:
Oh I'm so bad at checking notifications. Sorry! Taking a look now.

First error is:
```
We are unable to start your build at this time. You exceeded the number of users allowed for your plan. Please review your plan details and follow the steps to resolution.
```

I tried deactivating a few users to see if that helps, but it looks very much like it wants my credit card which I'm reluctant to do.

felixdivo, 2022-11-16:
I think [it should still work](https://docs.travis-ci.com/user/migrate/open-source-on-travis-ci-com/), but not quire sure. They aren't to clear on this.

We have/had [four things](https://github.com/hardbyte/python-can/blob/develop/.travis.yml) on travis
- test socketcan -> probably hard to replace
- arm64 builds -> ?
- building documentation -> easy to move to Github actions (isn't it alread moved [here](https://github.com/hardbyte/python-can/blob/develop/.github/workflows/build.yml#L112), so can't we delete it from travis?)
- deploy to PyPI -> easy to move

felixdivo, 2022-11-16:
I also think having too many open issues is also not helping to keep on overview over what is going on.

zariiii9003, 2022-11-16:
> I tried deactivating a few users to see if that helps, but it looks very much like it wants my credit card which I'm reluctant to do.

Yes, they want credit card info to avoid fake users/crypto miners. They will not charge you but i'd be hesitant too.

hardbyte, 2022-11-16:
Those last two items can be crossed off from changes in https://github.com/hardbyte/python-can/pull/1433

- building documentation - Indeed already migrated. I've just removed that part from travis config.
- deploy to pypi just moved

Would Qemu be an option for arm64 and maybe socketcan within docker?

felixdivo, 2022-11-16:
As far as I know, socketcan does not work within docker. Maybe this has changed but I doubt it.

hardbyte, 2022-11-16:
Oh that's right, I believe it can work but only when supported by the host kernel... which defeats the point in our case. I guess there is a small chance that vcan is now supported by the latest github worker images.

closed 2023-06-13 by zariiii9003