hardbyte/python-can #1222: Use can.logger only log received msg to file, but send msg can not logged

hongjojjay, 2022-01-19:
### Describe the bug
I use this below code, only received msg can save to hello.asc file, but send msg can not be saved. How to save the send msg?
tmplogger = can.Logger("hello.asc")
self.bus.add_listener(tmplogger )

### To Reproduce
None

### Expected behavior
None

### Additional context

OS and version: win10
Python version: 3.7.7
python-can version: 37.0.1
python-can interface/s (if applicable): pcan

zariiii9003, 2022-01-21:
Most interfaces support the `receive_own_messages` keyword for that.

felixdivo, 2022-01-21:
However, `pcan` does not seem to have that: https://python-can.readthedocs.io/en/develop/interfaces/pcan.html#bus

zariiii9003, 2022-01-21:
> However, `pcan` does not seem to have that: https://python-can.readthedocs.io/en/develop/interfaces/pcan.html#bus

Also VectorBus says this:
https://github.com/hardbyte/python-can/blob/d8e9dc5a64326926f744ecd2a9d0658a2acac5ed/can/interfaces/vector/canlib.py#L98
But BusABC does not mention `receive_own_messages`. Maybe we should add it to the BusABC arguments.

felixdivo, 2022-02-06:
I think we should close this issue (not a bug, just bad docs) and open a new one for improving documentation (and possibly the naming) of common parameters. Maybe we could also add a table comparing all the interfaces and their capabilities.

zhoulian1992, 2022-06-02:
Can you show me how to log receive message?  I don't know how to log can datas.

felixdivo, 2022-06-02:
@zhoulian1992 This has nothing to do with this issue. Have you tried `print(message)` or `str(message)`? If that does not work, open a new issue.