hardbyte/python-can #1227: Add mypy typing support

brent-stone, 2022-01-22:
### Problem
Downstream projects using mypy and other automated tooling need can.Message to have proper type hints

### Solution
Accept my pull request from brent-stone:mypy_support

### Alternatives
Gradually introduce modern (3.9+) typing into the python-can package.

brent-stone, 2022-01-23:
This is being implemented with the type hinting the dev branch. Closing.

closed 2022-01-23 by brent-stone

felixdivo, 2022-01-23:
Still thank you for your quick effort in #1226!

zariiii9003, 2022-01-23:
@felixdivo We might need to include a [py.typed](https://www.python.org/dev/peps/pep-0561/#id13) file so mypy works for downstream packages. But i'm not sure if it causes problems since `can.interfaces` is not really well annotated.