hardbyte/python-can #1242: Update CONTRIBUTERS.txt and history.rst

felixdivo, 2022-01-28:
Both files should be updated before the final release `4.0.0`.

Mentioned in  #1241.

felixdivo, 2022-02-04:
Help is welcome. While writing the `4.0.0rc0` release notes on GitHub, this neat list got created automatically (for the changes since the latest `3.x` release):

----

#### New Contributors

* @marcel-kanter made their first contribution in https://github.com/hardbyte/python-can/pull/562
* @bessman made their first contribution in https://github.com/hardbyte/python-can/pull/701
* @koberbe made their first contribution in https://github.com/hardbyte/python-can/pull/713
* @tamenol made their first contribution in https://github.com/hardbyte/python-can/pull/711
* @deonvdw made their first contribution in https://github.com/hardbyte/python-can/pull/731
* @ptoews made their first contribution in https://github.com/hardbyte/python-can/pull/739
* @chrisoro made their first contribution in https://github.com/hardbyte/python-can/pull/740
* @sou1hacker made their first contribution in https://github.com/hardbyte/python-can/pull/741
* @auden-rovellequartz made their first contribution in https://github.com/hardbyte/python-can/pull/762
* @typecprint made their first contribution in https://github.com/hardbyte/python-can/pull/764
* @tysonite made their first contribution in https://github.com/hardbyte/python-can/pull/780
* @Joey5337 made their first contribution in https://github.com/hardbyte/python-can/pull/812
* @Aajn made their first contribution in https://github.com/hardbyte/python-can/pull/789
* @josko7452 made their first contribution in https://github.com/hardbyte/python-can/pull/811
* @leventerevesz made their first contribution in https://github.com/hardbyte/python-can/pull/815
* @ericevenchick made their first contribution in https://github.com/hardbyte/python-can/pull/853
* @ixygo made their first contribution in https://github.com/hardbyte/python-can/pull/906
* @Gussy made their first contribution in https://github.com/hardbyte/python-can/pull/902
* @altendky made their first contribution in https://github.com/hardbyte/python-can/pull/920
* @philsc made their first contribution in https://github.com/hardbyte/python-can/pull/921
* @rliebscher made their first contribution in https://github.com/hardbyte/python-can/pull/926
* @jxltom made their first contribution in https://github.com/hardbyte/python-can/pull/905
* @kdschlosser made their first contribution in https://github.com/hardbyte/python-can/pull/946
* @tojoh511 made their first contribution in https://github.com/hardbyte/python-can/pull/955
* @s0kr4t3s made their first contribution in https://github.com/hardbyte/python-can/pull/956
* @jaesc made their first contribution in https://github.com/hardbyte/python-can/pull/968
* @NiallDarwin made their first contribution in https://github.com/hardbyte/python-can/pull/975
* @sdorre made their first contribution in https://github.com/hardbyte/python-can/pull/983
* @gordon-epc made their first contribution in https://github.com/hardbyte/python-can/pull/987
* @willson556 made their first contribution in https://github.com/hardbyte/python-can/pull/976
* @jjguti made their first contribution in https://github.com/hardbyte/python-can/pull/960
* @wiboticalex made their first contribution in https://github.com/hardbyte/python-can/pull/989
* @illuusio made their first contribution in https://github.com/hardbyte/python-can/pull/980
* @cperkulator made their first contribution in https://github.com/hardbyte/python-can/pull/761
* @simontegelid made their first contribution in https://github.com/hardbyte/python-can/pull/1042
* @DawidRosinski made their first contribution in https://github.com/hardbyte/python-can/pull/1056
* @fabiocrestani made their first contribution in https://github.com/hardbyte/python-can/pull/966
* @ChrisSweetKT made their first contribution in https://github.com/hardbyte/python-can/pull/1103
* @ausserlesh made their first contribution in https://github.com/hardbyte/python-can/pull/1116
* @wbarnha made their first contribution in https://github.com/hardbyte/python-can/pull/1110
* @projectgus made their first contribution in https://github.com/hardbyte/python-can/pull/1114
* @samsmith94 made their first contribution in https://github.com/hardbyte/python-can/pull/1142
* @alexey-tereshenkov-oxb made their first contribution in https://github.com/hardbyte/python-can/pull/1157
* @MattWoodhead made their first contribution in https://github.com/hardbyte/python-can/pull/1159
* @nbusser made their first contribution in https://github.com/hardbyte/python-can/pull/1128
* @domologic made their first contribution in https://github.com/hardbyte/python-can/pull/1140
* @fjburgos made their first contribution in https://github.com/hardbyte/python-can/pull/1119
* @pkess made their first contribution in https://github.com/hardbyte/python-can/pull/1187
* @felixn made their first contribution in https://github.com/hardbyte/python-can/pull/1144
* @Tbruno25 made their first contribution in https://github.com/hardbyte/python-can/pull/1204

hardbyte, 2022-02-04:
Well that's handy!

felixdivo, 2022-02-04:
Oh. I realize that I probably just pinged all of you by including that snipped. Anyways, thank you all for the great collaboration!

> Well that's handy!

Yes! One pretty much only needs to copy it over to the CONTRIBUTORS file.

hardbyte, 2022-02-04:
I'll remove email addresses from the CONTRIBUTORS file and add a link to https://github.com/hardbyte/python-can/graphs/contributors

Do you think adding github handles directly to `CONTRIBUTORS` is worthwhile?

PR #1251 (2022-02-07): Update history & contributors
Closes #1242
closed 2022-02-07 by hardbyte