hardbyte/python-can #1247: Unable to automatically find the Kernel headers. These are mandatory to build the pcan driver. Please set the KERNEL_LOCATION variable to the root directory of your own built Kernel.".  Stop.

TannerGilbert, 2022-01-31:
I'm currently using the [PCAN-USB](https://www.peak-system.com/PCAN-USB.199.0.html?&L=1) and the `python-can` library to read sensor values from my ESC using a code that look something like the following:

```python
# import the library
import can
can.rc['interface'] = 'pcan'
can.rc['channel'] = 'PCAN_USBBUS1'
can.rc['state'] = can.bus.BusState.PASSIVE
can.rc['bitrate'] = 1000000

bus = can.Bus()

# send a message
message = can.Message(arbitration_id=200)
bus.send(message, timeout=0.2)

# iterate over received messages
for msg in bus:
    speed = int.from_bytes(msg.data[:2], 'little', signed=True)
    motor_current = 0.1*float(int.from_bytes(msg.data[2:4], 'little', signed=True))
    temperature = 0.1*float(int.from_bytes(msg.data[4:6], 'little', signed=False))
    dc_voltage = 0.001*float(int.from_bytes(msg.data[6:8], 'little', signed=False))
```

On my local and remote Linux machines, that works fine, but when trying to run the PCAN driver installation in Docker, I'm getting the following error:

```
Makefile:131: *** "Unable to automatically find the Kernel headers. These are mandatory to build the pcan driver. Please set the KERNEL_LOCATION variable to the root directory of your own built Kernel.". Stop.
```

Dockerfile:

``` 
FROM python:3.9.6-slim-buster

RUN apt-get update -y && apt-get upgrade -y

RUN apt-get install -y make wget libusb-1.0-0 libusb-1.0-0-dev

RUN pip install pyserial==3.4 python-socketio[client]==5.5.1 python-engineio==4.3.1 requests==2.22.0 pyusb==1.1.1 PyJWT==2.0.1 python-can==3.3.4

RUN wget https://www.peak-system.com/fileadmin/media/linux/files/peak-linux-driver-8.14.0.tar.gz

RUN tar -xzf peak-linux-driver-8.14.0.tar.gz 

WORKDIR peak-linux-driver-8.14.0

RUN make clean 

RUN make 

RUN make install

WORKDIR /esc-control

COPY ./esc .

COPY ./serial_utils.py .

CMD ["python", "client.py"]
```

Is there any way of using the PCAN with the `python-can` library without installing the PCAN Linux drivers? 

Any help is greatly appreciated.

felixdivo, 2022-02-06:
Which part of the Dockerfile prints the error? Is it one of the `make` invocations? Then this isn't an issue with python-can itself but rather with PEAK's software not working in Docker (skipping `pip install [...] python-can[...]` should fail too).

hartkopp, 2022-02-06:
> Is there any way of using the PCAN with the `python-can` library without installing the PCAN Linux drivers?

The PCAN drivers are part of the Linux Kernel - there is usually no need to compile the out-of-tree PCAN Linux drivers (unless you need the chardev API).

Peak CAN Hardware usually runs out-of-the-box with the shipped CAN network (aka SocketCAN) drivers.

See their own website:
https://www.peak-system.com/fileadmin/media/linux/index.htm

TannerGilbert, 2022-02-10:
@felixdivo thanks for the reply. The error is occuring when running `make clean`.

TannerGilbert, 2022-02-10:
Also for more detail here is the `docker-compose.yml` I use:

```
version: '3'
services:
  esc:
    build:
      context: ./
      dockerfile: esc.Dockerfile
    restart: always
    volumes:
      - /dev:/dev
    privileged: true
```

hartkopp, 2022-02-10:
> @hartkopp when I'm trying to run PCAN without installing the drivers first I run into the following error:

>   File "/usr/local/lib/python3.9/site-packages/can/interfaces/pcan/pcan.py", line 172, in __init__
>     self.m_objPCANBasic = PCANBasic()
>   File "/usr/local/lib/python3.9/site-packages/can/interfaces/pcan/basic.py", line 363, in __init__
>     self.__m_dllBasic = cdll.LoadLibrary("libpcanbasic.so")

Is it possible that this is a Windows setup where you would need the 'pcan' library to access the PEAK hardware under Windows OS?

For Linux the CAN interfaces should use bustype socketcan, see:

https://python-can.readthedocs.io/en/master/interfaces/socketcan.html

When attaching the PEAK USB adapters to Linux a CAN interface like `can0` should appear automatically.

TannerGilbert, 2022-02-10:
@hartkopp I tested the above code on a Raspberry Pi 4 running Ubuntu and on my personal laptop running Pop OS. When plugging in the USB to CAN converter no `can` interface is created though. Can you think of any reason why this might be happening?

TannerGilbert, 2022-02-13:
@hartkopp the output of `dmesg` looks as follows:

```
[  761.262781] usb 1-1: new full-speed USB device number 6 using xhci_hcd
[  761.420025] usb 1-1: New USB device found, idVendor=0c72, idProduct=000c, bcdDevice=54.ff
[  761.420038] usb 1-1: New USB device strings: Mfr=10, Product=4, SerialNumber=0
[  761.420044] usb 1-1: Product: PCAN-USB
[  761.420048] usb 1-1: Manufacturer: PEAK-System Technik GmbH
[  761.540262] pcan: Release_20220124_n (le)
[  761.540267] pcan: driver config [mod] [isa] [pci] [pec] [usb] 
[  761.541221] pcan 1-1:1.0: PCAN-USB fw v8.4
[  761.591607] pcan: - usb device minor 32 found
[  761.591645] usbcore: registered new interface driver pcan
[  761.591649] pcan: major 508.
```

hartkopp, 2022-02-13:
> [  761.540267] pcan: driver config [mod] [isa] [pci] [pec] [usb] 
(..)
> [  761.591649] pcan: major 508.

You have installed the PEAK **character device** driver, which **blacklists** the original included Linux **netdevice drivers**.
Please uninstall the pcan driver =>  `make uninstall`

TannerGilbert, 2022-02-13:
@hartkopp Thanks a lot for your help. This fixed it for me. After uninstalling the PCAN driver I was able to get SocketCAN and the Docker Container to work.

closed 2022-02-13 by TannerGilbert