hardbyte/python-can #1248: TypeError: dict.get() takes no keyword arguments during USB2CAN initialization

wiboticalex, 2022-02-04:
### Describe the bug

When trying to use the usb2can interface, I get the following error:
TypeError: dict.get() takes no keyword arguments

### To Reproduce

```
can.ThreadSafeBus(
  interface="usb2can",
  channel="F8BAFED8",
  bitrate=500000,
)
```

Removing `d=` from line 106 in usb2canInterface.py seems to fix the issue and pass in the channel as the device_id as a positional argument.

### Expected behavior

The usb2can interface initializes without a TypeError. 

### Additional context

OS and version:  Windows 10 x64
Python version: Python 3.9.10
python-can version: 4.0.0 dev2 24ea2f2a4dbb6e72b41dc88646d0fc22ac1e10da
python-can interface/s (if applicable): usb2can

<summary>Traceback and logs</summary>

```
File "...\lib\site-packages\can\interface.py", line 120, in __new__
    bus = cls(channel, *args, **kwargs)
          │   │         │       └ {'bitrate': 500000}
          │   │         └ ()
          │   └ 'F8BAFED8'
          └ <class 'can.interfaces.usb2can.usb2canInterface.Usb2canBus'>

  File "...\lib\site-packages\can\interfaces\usb2can\usb2canInterface.py", line 106, in __init__
    device_id = kwargs.get("serial", d=channel)
                │      │                └ 'F8BAFED8'
                │      └ <method 'get' of 'dict' objects>
                └ {}

TypeError: dict.get() takes no keyword arguments
```

closed 2022-02-04 by hardbyte

felixdivo, 2022-02-04:
Thanks @wiboticalex for reporting!

As a side question, how did you get the nice traceback (that lists the values of the variables)?

wiboticalex, 2022-02-04:
The exception was generated using loguru with https://loguru.readthedocs.io/en/stable/api/logger.html#loguru._logger.Logger.exception

PR #1249 (2022-02-04): Fix guaranteed crash when using usb2can
Closes  #1248