hardbyte/python-can #1250: Please test the new 4.0.0rc0 pre-release - A lot has changed

felixdivo, 2022-02-04:
A new release candidate is available, [4.0.0rc0](https://github.com/hardbyte/python-can/releases/tag/4.0.0rc0)! It includes a ton of improvements from 2.5 years of development! :partying_face: You can install it with `pip install python-can==4.0.0rc0`.

**We need your help:** Please test the new release against your hardware and your own software and tools. If you experience problems (bugs, performance degradation, documentation issues, ...), please report them!

We will wait for at least a week for feedback. Depending on the number of issues that this brings up, we will consider releasing another *rc* build.

danielhrisca, 2022-02-08:
I get the following error when I replay an ASC file:

```
09:07:47  self = <can.io.asc.ASCReader object at 0x0000020898E659D0>
09:07:47  
09:07:47      def __iter__(self) -> Generator[Message, None, None]:
09:07:47          self._extract_header()
09:07:47      
09:07:47          for line in self.file:
09:07:47              line = line.strip()
09:07:47      
09:07:47              if not re.match(
09:07:47                  r"\d+\.\d+\s+(\d+\s+(\w+\s+(Tx|Rx)|ErrorFrame)|CANFD)",
09:07:47                  line,
09:07:47                  re.ASCII | re.IGNORECASE,
09:07:47              ):
09:07:47                  # line might be a comment, chip status,
09:07:47                  # J1939 message or some other unsupported event
09:07:47                  continue
09:07:47      
09:07:47              msg_kwargs: Dict[str, Union[float, bool, int]] = {}
09:07:47              try:
09:07:47                  _timestamp, channel, rest_of_message = line.split(None, 2)
09:07:47  >               timestamp = float(_timestamp) + self.start_time
09:07:47  E               AttributeError: 'ASCReader' object has no attribute 'start_time'
```

the error occurs [here](https://github.com/hardbyte/python-can/blob/develop/can/io/asc.py#L225)

felixdivo, 2022-02-18:
> I don't see a need for another RC release and will release 4.0.0 soon if no one complains. ;)

No one did, so I'll consider this done. Will add a changelog and push a release.

closed 2022-02-18 by felixdivo