hardbyte/python-can #1253: XL_ERR_QUEUE_IS_FULL when sending large data (Vector hardware)

dpatel20, 2022-02-05:
### Describe the bug
I am using udsoncan to run some diagnostics, including 'Transfer Data'. As part of this, I send 4kB packets using transfer data of UDS (utilising python-can-isotp via udsoncan).

Randomly I get XL_ERR_QUEUE_IS_FULL whilst python-can is part way through.

What I **believe** is happening is that ```xlCanTransmit()``` is being called before it has finished sending the previous message. I've checked the python CAN isotp code and cannot see anywhere that it might be sending a second call to transmit before the previous one has finished.

Only seems to be a problem on a multi-frame message.

Whilst I was debugging, I put a print message in the ```xlCanTransmit()``` and the problem did not occur. I know that the `print` will cause a delay and, I presume, that delay is enough to stop the issue. In fact, I can even do something similar in the python-can-isotp `send` and it has the same effect - i.e. putting a `print` (delay) anywhere in the send chain seems to be sufficient.

Does ```xlCanTransmit()``` not block until the current frame is transmitted? I presume the same issue could occur if two threads were trying to send on the same connection??

Perhaps I am missing a parameter to set?

```python
# This the setup. Adding in isotp_params doesn't stop the issue.
bus = VectorBus(channel=0, bitrate=500000)
tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x7B5, rxid=0x7BD)
stack = isotp.CanStack(bus=bus, address=tp_addr)
conn = PythonIsoTpConnection(stack) 

```

### Additional context

OS and version: Windows 11
Python version: 3.9
python-can version: Tried with 3.4 & latest 4.0rc
python-can interface/s (if applicable): Vector CANcaseXL

dpatel20, 2022-02-06:
According to [this page](https://support.vector.com/kb?id=kb_article_view&sysparm_article=KB0011439&sys_kb_id=663313f687b33c10b9f233770cbb3582&spa=1):

> You have a program that writes messages very quickly to the buffer so that the data rate of the real bus is exceeded (e.g. while-/for-loops).

Suggesting that the bus data rate needs to be respected to avoid this error. 

Also, this [comment ](https://github.com/rbei-etas/busmaster/blob/de29ec1514558b3b7e0504e07d38ce0d65616d0f/Sources/BUSMASTER/CAN_Vector_XL/CAN_Vector_XL.cpp#L1805) suggests that the data is placed into a buffer (USB case) and returns - so it doesn't wait until message is actually placed on the bus before returning.

Can anyone suggest what might the appropriate way to respect the data rate of the bus?

isaacag, 2024-10-08:
I was able to come across this issue as well, this is the setup:

CAN-FD timing:
f_clock=80_000_000,
nom_bitrate=500_000,
nom_sample_point=80,
data_bitrate=2_000_000,
data_sample_point=80

Hardware: Vector VN1630A and VN1610

Bus: 90 messages, cycle time 1/3 of them 10ms, 1/2 of them < 50ms

XL_ERR_QUEUE_IS_FULL error creates a chain reaction that kills most periodic message threads, this work-around overriding one of the VectorBus methods is working:

```python
def _send_can_fd_msg_sequence(self, msgs: Sequence[Message]) -> int:
    """Send CAN FD messages and return number of successful transmissions."""
    mask = self._get_tx_channel_mask(msgs)
    message_count = len(msgs)
    xl_can_tx_event_array = (xlclass.XLcanTxEvent * message_count)(*map(self._build_xl_can_tx_event, msgs))
    msg_count_sent = ctypes.c_uint(0)
    try:
        self.xldriver.xlCanTransmitEx(self.port_handle, mask, message_count, msg_count_sent, xl_can_tx_event_array)
        time.sleep(0.001)
    except VectorOperationError as e:
            logger.warning(e)
    return msg_count_sent.value
```

Concern here is, by doing this we are somehow compromising the performance of the bus and the accuracy of the cycle times.