hardbyte/python-can #1254: Unable to receive correct data

Omkar-KAN, 2022-02-07:
### Describe the bug
Not receiving correct data from can interface

### To Reproduce
Send a single message through kvaser can channel 0 using 
msg = can.Message(arbitration_id=0x131, dlc = 6, data=[1, 2, 3, 12, 10, 11], is_error_frame=False, is_extended_id=False)
bus.send(msg)

Using kvaser can channel 1 interface
for msg in bus
  Print (msg)

Same as the documentation 

### Expected behavior
We should be receiving dlc = 6 data = 1 2 3 12 10 11
I read dlc =4  Error frame data = 20 28 00 00 (irrespective of what data is being sent)

### Additional context

OS and version: win10
Python version: py3.10.2
python-can version: 3.3.4
python-can interface/s (if applicable): kvaser, bitrate 500000

This also happens if you send data through pcan pci bus and receive data on kvaser. Doesn't work the other way around.

felixdivo, 2022-02-07:
Are the channels connected, i.e. are you expecting to read the data you send one the first from the second? Is this the same `bus` instance?

You could also use the most recent version and check if the problem is already fixed with `pip install python-can==4.0.0rc0`.

MunchZkin, 2022-02-07:
Hey! Sorry I did not format that bug report properly, Thank you very much!
I tried upgraded python-can to the most recent version but I still do not have any luck.
To test this can interface I am using [Kvaser USBcan Light 2xHS](https://www.kvaser.com/product/kvaser-usbcan-light-2xhs/) and connected the high and low respectively. 

To send a single message I use 
`import can`
`bus = can.interface.Bus(bustype='kvaser', channel=0, bitrate=500000)`
`bus.flush_tx_buffer()`
`print("Flushed!!!")`
`print(bus)`
`msg = can.Message(arbitration_id=0x131, dlc = 6, data=[1, 2, 3, 12, 10, 51], is_error_frame=False, is_extended_id=False)`
`print(msg)`
`bus.send(msg)`

For which I get the following output; 
Flushed!!!
Kvaser USBcan Light 2xHS, S/N 17161 (#1)
Timestamp:        0.000000        ID: 0131    S Rx                DL:  6    01 02 03 0c 0a 33

and to receive using this code
`import can`
`bus = can.interface.Bus(bustype='kvaser', channel=1, bitrate=500000)`
`print(bus)`
`for msg in bus:`
    `print(msg)`
and for this I get the following output    
Kvaser USBcan Light 2xHS, S/N 17161 (#2)
(can.Message(timestamp=1644261029.5125065, arbitration_id=0x0, is_extended_id=False, is_error_frame=True, channel=0, dlc=4, data=[0x0, 0x8, 0x0, 0x0]), False)
Timestamp: 1644261029.512587        ID: 0000    S Rx E              DL:  4    00 08 00 00                 Channel: 0
Timestamp: 1644261029.512587        ID: 0000    S Rx E              DL:  4    00 08 00 00                 Channel: 0
Timestamp: 1644261029.512676        ID: 0000    S Rx E              DL:  4    00 08 00 00                 Channel: 0
Timestamp: 1644261029.512676        ID: 0000    S Rx E              DL:  4    00 08 00 00                 Channel: 0
Timestamp: 1644261029.512767        ID: 0000    S Rx E              DL:  4    00 08 00 00                 Channel: 0
Timestamp: 1644261029.512846        ID: 0000    S Rx E              DL:  4    00 08 00 00                 Channel: 0
Timestamp: 1644261029.512846        ID: 0000    S Rx E              DL:  4    00 08 00 00                 Channel: 0
Timestamp: 1644261029.512937        ID: 0000    S Rx E              DL:  4    00 08 00 00                 Channel: 0
Timestamp: 1644261029.513016        ID: 0000    S Rx E              DL:  4    00 08 00 00                 Channel: 0
Timestamp: 1644261029.513016        ID: 0000    S Rx E              DL:  4    00 08 00 00                 Channel: 0
Timestamp: 1644261029.513107        ID: 0000    S Rx E              DL:  4    00 48 00 00                 Channel: 0
Timestamp: 1644261029.513186        ID: 0000    S Rx E              DL:  4    00 48 00 00                 Channel: 0
Timestamp: 1644261029.513277        ID: 0000    S Rx E              DL:  4    00 48 00 00                 Channel: 0
Timestamp: 1644261029.513277        ID: 0000    S Rx E              DL:  4    00 48 00 00                 Channel: 0
Timestamp: 1644261029.513356        ID: 0000    S Rx E              DL:  4    20 48 00 00                 Channel: 0
Timestamp: 1644261029.513356        ID: 0000    S Rx E              DL:  4    20 48 00 00                 Channel: 0
Timestamp: 1644261029.513447        ID: 0000    S Rx E              DL:  4    20 48 00 00                 Channel: 0
and keeps on repeating the last line for a few seconds and stops. 

From my understanding the can settings must be reset?
like bus.stop()

Omkar-KAN, 2022-02-09:
> No problem with the formatting. A bus is best used as a context manager like [here](https://github.com/hardbyte/python-can/blob/develop/examples/send_one.py#L15), as this will always call `bus.shutdown()` for you, even on exceptions being raised. And the `bus.flush_tx_buffer()` should not be required.
> 
> It is surprising that you receive multiple messages while you only send a single one. If you just listen, and do not run your first script which sends, you receive nothing, right?

I tried the exact same script you shared to send a single message, but I still received the same wrong message. I also tried to send data using PCAN PCI but no luck! and when I reversed the Rx Tx I received `Bus error: an error counter reached the 'heavy'/'warning' limit`

Regarding receiving multiple messages, yes I do not receive any messages on the listener until any data is transmitted over the bus interface. And for every message I send, I receive roughly 409 messages. 

I have triple checked again this time with the channels including virtual channels created by 'kvaser' to make sure I'm using the right ones.

felixdivo, 2022-02-09:
Uhm, do I correctly assume @Omkar-KAN and @MunchZkin, you are the same person? ^^

Omkar-KAN, 2022-02-09:
> Uhm, do I correctly assume @Omkar-KAN and @MunchZkin, you are the same person? ^^

Yes, That's me

Nangsaga, 2024-03-27:
Hello,
I had the same issue when using socketcan interface. Please share any information if the issue was solved.
Many thanks.

MunchZkin, 2024-03-27:
We had to use a male to male adapter to connect the hardware to the can pcie, I was under the impression there was a terminal resistance, but it wasn't. This might be the same with you?
You can google how to wire up this resistance, as it has been a while I have worked on it now

Nangsaga, 2024-03-27:
> We had to use a male to male adapter to connect the hardware to the can pcie, I was under the impression there was a terminal resistance, but it wasn't. This might be the same with you? You can google how to wire up this resistance, as it has been a while I have worked on it now

Thank for reply. I checked the resistance between CAN-L and CAN-H terminals of my CAN PCI-E and it was approx. 60 ohm. This means there are two 120ohm resistances at the both and of the bus and I think I do not have problem of hardware connections.