hardbyte/python-can #1255: 4.0.0-rc blf conversion

Rampagy, 2022-02-07:
### Describe the bug

If I convert a .blf to .asc and then back to .blf my expectation is that the original and resulting blf would be very similar in size.  Version 3.3.4  produces a blf that is much more similar in size than what 4.0.0 does.  What is all the extra data?

Version 3.3.4:
[image]

Version 4.0.0:
[image]

### To Reproduce

1. Convert a large blf file (>= 5MB) to an asc 
    *  For some reason this issue doesn't occur on smaller files for me
2. Convert the resulting asc back to blf
3. Compare the original blf to the resulting blf

### Expected behavior

The original and resulting blf should be very similar in size.

### Additional context

OS and version:
Python version: 3.9.6
python-can version: 4.0.0-rc
python-can interface/s (if applicable): Vector Hardware

<details><summary>Traceback and logs</summary>

```python
import can
import filecmp

FILE_NAME = 'example'

# read in BLF file
msgIterator = can.BLFReader(FILE_NAME + '.blf')

# convert BLF to .asc
with can.ASCWriter(FILE_NAME + '.asc') as asc_writer:
    for msg in msgIterator:
        asc_writer.on_message_received(msg)

# read in ASC file
msgIterator = can.ASCReader(FILE_NAME + '.asc')

# convert ASC back to .blf
with can.BLFWriter(FILE_NAME + '_0.blf') as blf_writer:
    for msg in msgIterator:
        blf_writer.on_message_received(msg)

if filecmp.cmp(FILE_NAME + '.blf', FILE_NAME + '_0.blf'):
    print('Files are the same!')
else:
    print('Files are not the same!')
```
</details>

christiansandberg, 2022-02-08:
The default compression level has changed from max (9) to the default of zlib (6 I think). This is because while 9 gives slightly smaller files it is much slower, so for most applications it is not worth it. But you can still pass that as an argument to BLFWriter.

Rampagy, 2022-02-08:
I tried and now the results match my expectation!

closed 2022-02-08 by Rampagy