hardbyte/python-can #1261: Notifiers don't seem to work with virtual busses in v4.0.0

andlaus, 2022-02-19:
### Describe the bug

Notifiers don't seem to work with virtual busses in v4.0.0. At least we see some failed tests in [cantools](https://github.com/cantools/cantools): https://github.com/cantools/cantools/runs/5259626336 .

### To Reproduce

Create a notifier for a virtual bus, or run `test/test_tester.py` from [cantools](https://github.com/cantools/cantools)

### Expected behavior

no raised exception in the notifier code if the bus object does not exhibit a `fileno` attribute.

### Additional context

OS and version: Linux 5.16.4
Python version: tested on 3.8, 3.9
python-can version: 4.0.0
python-can interface/s (if applicable): virtual

backtrace:
```
Traceback (most recent call last):
  File "/home/runner/work/cantools/cantools/tests/test_tester.py", line 271, in test_bad_message_name
    tester, _ = setup_tester('Node1')
  File "/home/runner/work/cantools/cantools/tests/test_tester.py", line 61, in setup_tester
    tester = cantools.tester.Tester(dut_name,
  File "/home/runner/work/cantools/cantools/cantools/tester.py", line 295, in __init__
    self._notifier = can.Notifier(can_bus, [listener])
  File "/opt/hostedtoolcache/Python/3.9.10/x64/lib/python3.9/site-packages/can/notifier.py", line 59, in __init__
    self.add_bus(each_bus)
  File "/opt/hostedtoolcache/Python/3.9.10/x64/lib/python3.9/site-packages/can/notifier.py", line 69, in add_bus
    reader = bus.fileno()
AttributeError: 'CanBus' object has no attribute 'fileno'
```

zariiii9003, 2022-02-20:
Hello,

you are not subclassing from BusABC here: https://github.com/cantools/cantools/blob/196a1e4e4cee7c4d8255c5c399e3cc20528e78d9/tests/test_tester.py#L16

That's why you are getting an AttributeError for `fileno()`

andlaus, 2022-02-20:
That fixed it, thanks! The error is a bit enigmatic IMO, but I don't see what python-can could do about this. Closing issue...

closed 2022-02-20 by andlaus